PRBonn/kiss-icp #165: Deterministic poses

benemer, 2023-05-21:
Moved from https://github.com/PRBonn/kiss-icp/issues/151#issuecomment-1539669538

> Hi, again, I would like to ask if KISS-ICP is deterministic, as I obtained slightly different results on the same sequence. I wonder if there is a way to fix the randomness or am I making a mistake? This would be great for debugging, since KISS-ICP is the very first step of a complicated framework.

benemer, 2023-05-21:
Are you using ROS or Python? How much different are the poses from different runs on the same data?

Using the single thread implementation should give more similar results across runs. Maybe @nachovizzo can comment on this.

nachovizzo, 2023-05-25:
All the slightly non-deterministic behaviors we observed where due to multithreading. Can you please check the single thread implementation and see if the problems disappears?

eugeniu1994, 2023-05-25:
Thank you @nachovizzo for so quick reply.
I've tested it without the TBB library, so no multithreaded is involved. Also, I tried to avoid using the adaptive threshold, so I set the fixed sigma:
//const double sigma = GetAdaptiveThreshold();
const double sigma = 0.5; //make it fixed
this still does not solve the issue.
A similar plot to show the number of points in the local map.
[image]

I suspect that the randomness might be introduced by Eigen ldlt() decomposition of the Hessian matrix in GN estimation, but I have to test it and I'll let you know. Other than that I have no idea what might cause this randomness.

nachovizzo, 2023-05-25:
@eugeniu1994 have you checked #152 ?

If you are running the ROS node, there might be a non-deterministic behavior based on your machine (ros is just replaying the bagfile). Another option is to run the bagfile with a "slower" clock. In any case, "not a kiss-icp problem"

You could also do the same plot, but with the RAW pointclouds that ROS is giving, just to make sure that the problem is not at the core of KISS-ICP and maybe a "ROS-related" issue.

nachovizzo, 2023-06-19:
@samirangode have you tried again after #162 got merged?

nachovizzo, 2023-07-13:
Closing thing due 3 weeks of inactivity

closed 2023-07-13 by nachovizzo

benemer, 2024-04-26:
If this is still an issue, https://github.com/PRBonn/kiss-icp/pull/300 should solve it (apart from the abovementioned effects from ROS).