PRBonn/kiss-icp #167: Using result in of poses_kitti.txt leading to strange results

SvenMuc, 2023-05-25:
Hi,
I'm trying to use the kiss-icp to generate a poses.txt file for the [point_labeler](https://github.com/jbehley/point_labeler) tool.
The kiss-icp poses_kitt.txt format seem to be correct, and can be read directly into the labeler tool. But the results are looking different. 

**Kiss-ICP Results**
The results are looking pretty good.
[image]

**Point Labeler Results using the Kiss-ICP poses**
There seems to be mainly an issue with the heading.
[image]

Any ideas? Are the poses formats different?
Are there any further conversions required?

nachovizzo, 2023-05-25:
@SvenMuc, what is the "results poses with the point labeler"? I haven't used this myself, so maybe @jbehley can shed some light on it

SvenMuc, 2023-05-25:
Maybe my description was a bit misleading. I applied the following steps:

1. Generate poses_kitt.txt with kiss-icp. According the kiss-icp visualization the match looks pretty accurate.
2. Use the poses_kitti.txt directly in point_labeler to map the single scans to one point cloud again. 

Both tools use the same poses format, therefore I would expect a similar accuracy in the point cloud visualizations.
I used exactly the same Lidar scans (dataset) in both tools.