PRBonn/kiss-icp #170: Point Cloud pre-processing required?

SvenMuc, 2023-05-26:
Hello,

I'm wondering, if I need to pre-process my point cloud in order to make kiss-icp export the 6DOF poses correctly into the kitti format.

The point cloud is provided in the sensor frame.
The sensor is mounted in a certain height (2.5 m) and with a pitch angle (~16 deg).

The matching in the Open3D visualization looks perfect.

**Top View**
[image]

**Side View**
[image]

But in the exported poses I see an increasing height tz.
Everything seems to be pitched.

[image]

[image]

nachovizzo, 2023-06-05:
@SvenMuc this looks interesting 🤔

Would you mind sharing the data so we can test it in house?

benemer, 2023-06-06:
How large is the data? If possible, you could upload it to a cloud storage space (Google Drive, Dropbox) and send us the link via email. You can find our contact details [here](https://www.ipb.uni-bonn.de/people/). Thanks!

SvenMuc, 2023-06-06:
The zip File is 25 GBs which is currently not supported by an free storage spaces.
I think I need to cut the rosbag into a smaler sequence.

SvenMuc, 2023-06-07:
@benemer and @nachovizzo I send you a PM with the data.

closed 2023-06-13 by benemer