PRBonn/kiss-icp #174: In which frame has the laser scan data to be

maxkremmel, 2023-06-16:
Hi all, 
The tf's of my robot look as follows:
[image]

As you can see, the tf of the laser scan (velodyne) is above and behind the base_link. 
As a result, kiss spits out the odometry in the velodyne frame (you can see that in the following GIF). What do I need to change or to do to get the LiDAR odometry in the base_link frame?
[image]

Do I have to convert the lidar data first into the base_link frame?

Thanks in advance.
Max

nachovizzo, 2023-06-19:
Just tell the ros node your child frame, for example, `child_frame:=base`. Maybe related to #92

closed 2023-06-19 by nachovizzo

nachovizzo, 2023-10-03:
Please check if this fix your issue https://github.com/PRBonn/kiss-icp/pull/232 

Apologize for the delay in the fix!

PR #232 (2023-10-27): Add support for multi-sensor scenarios
# Description

These are the required changes to use KISS-ICP as an odometry source for multi-sensor scenarios. Although the ticket is 
about the comparison, to have the odometry in the `base_footprint` coordinate frame, we need these changes.

This PR also relates to and fixes #174 

# Tf tree fix

So the tf logic was a total disaster, and I hope to fix it with this PR. The basic idea is as follows.

- If the user requires a particular `base_frame`, in our case, `base_footprint` (but it could be `base_link`), then all the point clouds must be expressed i [... cut]