PRBonn/kiss-icp #179: Pydantic-settings broke kiss_icp Python, due to it's major version update to 2.0.1

bexcite, 2023-07-05:
Seems across all OSs the same, just installing `kiss-icp` from PyPi and running `kiss_icp_pipeline` makes a crash and info about updating code to accommodate Pydantic-settings update changes :(

Fresh venv, Python3.9, Mac M1 Ventura (and also Python3.8, Ubuntu 22.04):
```
pip list | grep kiss
kiss-icp          0.2.9
```

Installation logs (partial):
```
pip install kiss-icp
Looking in indexes: https://pypi.org/simple, 
Collecting kiss-icp
  Using cached kiss_icp-0.2.9-cp39-cp39-macosx_11_0_arm64.whl (307 kB)
Collecting natsort (from kiss-icp)
  Using cached natsort-8.4.0-py3-none-any.whl (38 kB)
Collecting numpy (from kiss-icp)
  Downloading numpy-1.25.0-cp39-cp39-macosx_11_0_arm64.whl (14.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.0/14.0 MB 605.4 kB/s eta 0:00:00
Collecting plyfile (from kiss-icp)
  Using cached plyfile-0.9-py3-none-any.whl (30 kB)
Collecting pydantic (from kiss-icp)
  Downloading pydantic-2.0.1-py3-none-any.whl (358 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 358.4/358.4 kB 1.0 MB/s eta 0:00:00
```

Run and crash:
```
kiss_icp_pipeline ~/data/sample-data/OS-1-128_v3.0.1/OS-1-128_v3.0.1_1024x10_20230216_142857-000.pcap
╭─────────────────────────────── Traceback (most recent call last) ────────────────────────────────╮
│ /Users/pavlo.bashmakov/code/swt/.venv-kiss-temp1/lib/python3.9/site-packages/kiss_icp/tools/cmd. │
│ py:217 in kiss_icp_pipeline                                                                      │
│                                                                                                  │
│   214 │                                                                                          │
│   215 │   # Lazy-loading for faster CLI                                                          │
│   216 │   from kiss_icp.datasets import dataset_factory                                          │
│ ❱ 217 │   from kiss_icp.pipeline import OdometryPipeline                                         │
│   218 │                                                                                          │
│   219 │   OdometryPipeline(                                                                      │
│   220 │   │   dataset=dataset_factory(                                                           │
│                                                                                                  │
│ ╭─────────────────────────────────────────── locals ───────────────────────────────────────────╮ │
│ │          config = None                                                                       │ │
│ │            data = PosixPath('/Users/pavlo.bashmakov/data/sample-data/OS-1-128_v3.0.1/OS-1-1… │ │
│ │      dataloader = 'ouster'                                                                   │ │
│ │ dataset_factory = <function dataset_factory at 0x105c11040>                                  │ │
│ │          deskew = False                                                                      │ │
│ │            jump = 0                                                                          │ │
│ │       max_range = None                                                                       │ │
│ │            meta = None                                                                       │ │
│ │         n_scans = -1                                                                         │ │
│ │        sequence = None                                                                       │ │
│ │           topic = None                                                                       │ │
│ │         version = None                                                                       │ │
│ │       visualize = False                                                                      │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                                  │
│ /Users/pavlo.bashmakov/code/swt/.venv-kiss-temp1/lib/python3.9/site-packages/kiss_icp/pipeline.p │
│ y:33 in <module>                                                                                 │
│                                                                                                  │
│    30 import numpy as np                                                                         │
│    31 from pyquaternion import Quaternion                                                        │
│    32                                                                                            │
│ ❱  33 from kiss_icp.config import load_config, write_config                                      │
│    34 from kiss_icp.kiss_icp import KissICP                                                      │
│    35 from kiss_icp.metrics import absolute_trajectory_error, sequence_error                     │
│    36 from kiss_icp.tools.pipeline_results import PipelineResults                                │
│                                                                                                  │
│ ╭─────────────────────────────────────────── locals ───────────────────────────────────────────╮ │
│ │ contextlib = <module 'contextlib' from                                                       │ │
[... 60 lines cut ...]
│                                                                                                  │
│ /Users/pavlo.bashmakov/code/swt/.venv-kiss-temp1/lib/python3.9/site-packages/pydantic/_migration │
│ .py:279 in wrapper                                                                               │
│                                                                                                  │
│   276 │   │   │   # skip the warning here because a deprecation warning will be raised elsewhe   │
│   277 │   │   │   return import_string(DEPRECATED_MOVED_IN_V2[import_path])                      │
│   278 │   │   if import_path == 'pydantic:BaseSettings':                                         │
│ ❱ 279 │   │   │   raise PydanticImportError(                                                     │
│   280 │   │   │   │   '`BaseSettings` has been moved to the `pydantic-settings` package. '       │
│   281 │   │   │   │   f'See https://docs.pydantic.dev/{VERSION}/migration/#basesettings-has-mo   │
│   282 │   │   │   │   'for more details.'                                                        │
│                                                                                                  │
│ ╭────────────────────────────── locals ───────────────────────────────╮                          │
│ │         import_path = 'pydantic:BaseSettings'                       │                          │
│ │              module = 'pydantic'                                    │                          │
│ │                name = 'BaseSettings'                                │                          │
│ │ PydanticImportError = <class 'pydantic.errors.PydanticImportError'> │                          │
│ ╰─────────────────────────────────────────────────────────────────────╯                          │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
PydanticImportError: `BaseSettings` has been moved to the `pydantic-settings` package. See https://docs.pydantic.dev/2.0.1/migration/#basesettings-has-moved-to-pydantic-settings for more details.

For further information visit https://errors.pydantic.dev/2.0.1/u/import-error
```

In general for better stability it would be good to set dependencies ranges in `setup.py` for `install_requires` list, at least restrict to the current major versions, otherwise it will continue happening and braking under us by any third party deps major bump.

nachovizzo, 2023-07-05:
I'll fix this ASAP. Thanks for reporting again @bexcite!

I was always a bit allergic to put "random" versions on the requirements, since in the past I found myself struggling to get a very specific version of python (as an example) for no good reason. It's true that at some point this will always break, as you know.

I'll fix the pydantic one and meditate about the possibility of fixing the other dependencies. If you have any thoughts on this please feel free to share it :)

bexcite, 2023-07-05:
One thing you can do is to limit the upper bound, like add `<X` parts for every package, which should be fine to restrict pulling the major packages updates (that's usually the most troubling and huge changes, like that pydantic thing).

Random example from elsewhere for some ideas:
```
    install_requires=[
        'psutil >=5.9.5, <6',
        'click >=8.1.3, <9',
        'requests >=2.0, <3',
        'numpy >=1.19, <2, !=1.19.4',
        # scipy is not supported on Mac M1 with Mac OS < 12.0
        'scipy >=1.7, <2;platform_system != "Darwin" or platform_machine != "arm64" or platform_version >= "21.0.0"',
    ],
```

closed 2023-07-05 by nachovizzo

nachovizzo, 2023-07-05:
@bexcite I just pushed a new python package 👌 please let me know if there is any issue. Thanks

bexcite, 2023-07-05:
@nachovizzo hooray! thanks for quick fix, it works and it helps a lot!