PRBonn/kiss-icp #200: Deskewing not working for certain lidars

stm32f303ret6, 2023-08-02:
undistortion (ego motion compensation) doesn't work with livox lidar? in OdometryServer.cpp it looks for pointfields of "t", "time" or "timestamp" but none of them are found in the PointCloud2 message that publishes the ros2 livox driver, therefore the vector size of the timestamps is 0 https://github.com/PRBonn/kiss-icp/blob/main/ros/ros2/OdometryServer.cpp#L126

stm32f303ret6, 2023-08-02:
Hola Nacho! 

The PointCloud2 message that comes from the driver does not have a timestamp, but I just checked the livox ros2 driver repository and apparently in new versions they added timestamp to the message! So I will update the repository and then I'll tell you, thanks!

nachovizzo, 2023-08-02:
Ok thanks! Feel free to comment on this issue

closed 2023-08-02 by nachovizzo