PRBonn/kiss-icp #209: Is there an assumption in deskewing that the lidar scan is input continuously?

jaeyoungjo99, 2023-08-18:
Hello I'm Jaeyoung.
Thank you for your great works.

I'm currently working on lidar odometry using vertical object feature points in urban scenariao.

[image]

I have question in point deskewing parts.
As far as I understand the code, your code normalizes the time stamp of the points so that timestamp can be expressed in relative time.

But in Deskew.cpp, The delta pose is obtained through the amount of pose movement from N-2 to N-1, and it seems that delta_pose is relatively applied from the point of the first timestamp to the point of the last timestamp of the current point cloud.

https://github.com/PRBonn/kiss-icp/blob/f078513c92933d24b309e2853148b0ac06bbd414/cpp/kiss_icp/core/Deskew.cpp#L44

[image]

As shown in the figure, if points are not completely formed in the point cloud from scan start to scan end, the phenomenon of excessive deskewing of points occurs.

Also, for example, after obtaining LIDAR at 20Hz, the above phenomenon occurs when data is input to Kiss icp at a cycle of 10Hz. In that case, the bias of the odometry increases, especially when the vehicle turns.

In this scenario, when I manually fixed  code like below, this was fixed.
`const auto motion = Sophus::SE3d::exp((timestamps[i] - mid_pose_timestamp)*0.5 * delta_pose);`

[image]

I wonder if there is a way to compensate for the phenomenon through other methods. Or, if it was designed like this to simplify the code, I will personally modify the code and use it.

Thanks for the nice code and paper.

benemer, 2023-08-18:
Hey!

The de-skewing works as expected, the issue is the normalization in ROS, see [here](https://github.com/PRBonn/kiss-icp/blob/f078513c92933d24b309e2853148b0ac06bbd414/ros/ros1/Utils.hpp#L81). Currently, we expect to receive  timestamps in the range from the start of the scan to the end of the scan. 

In your described example, if you only received points for the first half of the duration of the scan, the points should have a timestamp in the range [0,0.5] without further normalization. Your fix is doing a similar thing, although it also scales the `mid_pose_timestamp` which should not be the case. 

Could you please remove the normalization and try again?

```diff
diff --git a/ros/ros1/Utils.hpp b/ros/ros1/Utils.hpp
index 137b5bc..b18d8e0 100644
--- a/ros/ros1/Utils.hpp
+++ b/ros/ros1/Utils.hpp
@@ -78,7 +78,7 @@ inline auto ExtractTimestampsFromMsg(const PointCloud2::ConstPtr msg, const Poin
         for (size_t i = 0; i < n_points; ++i, ++it) {
             timestamps.emplace_back(static_cast<double>(*it));
         }
-        return NormalizeTimestamps(timestamps);
+        return timestamps;
     };
 
     // Get timestamp field that must be one of the following : {t, timestamp, time}
```

Best
Benedikt

jaeyoungjo99, 2023-08-19:
@benemer 

Thanks for the reply. 

In my current scenario, point clouds come in at intervals of 0.1 seconds, but the relative time of the points inside each point cloud is formed between 0.0 and 0.05 seconds.
In this case, when normalization is not used, it is thought that delta_pose will be applied from 0 to 0.05 instead of from 0 to 0.5.

So I think the above problem will be solved if the time difference information from N-2 to N-1 is considered in the DeSkewScan function.

And can you explain the mid_pose_timestamp variable?

I am always inspired by kiss-icp and its paper.

Thank you.

benemer, 2023-08-21:
Ah I see! I thought you were already feeding normalized timestamps, meaning between 0 and 0.5. In that case, you would have to write your own normalization instead of using ours.

You are right, the way to solve this more generically would be to additionally use the time information of the two scans and then normalize the timestamps with respect to these. If you would like to work on this and contribute to the project, feel free to open a pull request :)

We linearly interpolate the motion between the two frames and transform each point to the (also interpolated) reference frame of the vehicle at the `mid_pose_timestamp`. We chose 0.5 because we found out that this reference was also used by the KITTI people to de-skew the raw point clouds for the benchmark. However, this does not affect the relative estimation between frames (because it is just a shift of reference).

Best,
Benedikt

closed 2023-08-21 by nachovizzo

jaeyoungjo99, 2023-08-21:
Thank you for your replying!

I'll try to understand your code more, find improvements and make a pull request :)

Thanks