PRBonn/kiss-icp #213: Map based localization

InguChoi, 2023-08-28:
Hello, thank you for sharing your research!

I have a question that is it possible to localize the robot`s pose with existing .pcd map?

Thank you!

nachovizzo, 2023-08-28:
Hello @InguChoi,

KISS-ICP is an odometry pipeline, and as such, we don't support map-based localization (#90, #147). Nevertheless, you can use the pipeline for post-tracking on a given pre-existing map, but you would have to modify the source code quite a bit.

For to moment, we don't plan to implement such functionality. Thanks for your interest in the project.

closed 2023-08-28 by nachovizzo

InguChoi, 2023-08-28:
Thank you for fast anwser!

DeepDuke, 2024-04-29:
Sorry, I want to ask if the current KISS-ICP supports localization given a prior map? @nachovizzo

nachovizzo, 2024-05-01:
@DeepDuke we don't but you could check out Florian's amazing work where he extended to do it 

Paper: http://arxiv.org/abs/2311.01823 Code:
https://github.com/TUMFTM/FlexMap_Fusion (ROS 2 node designed to work with #Autoware. Handed in at the Autoware Challenge 2023)