PRBonn/kiss-icp #215: Chaotic Odometry using Livox HAP (ROS1 and Python)

PlayWeird, 2023-08-29:
[image]

I'm really excited about the possibilities of kiss-icp, but as of yet, I've been unable to get satisfactory results using the Livox HAP lidar. I got similar results using the ROS1 and pip methods. My .bag recording was straightforward, with slight movement in an indoor environment; I'm happy to provide the .bag file if you are interested. I've only tried this with your default parameters so far. I plan to try this package using various Velodyne lidar in the near future and will update if I get better results. 

Thanks

nachovizzo, 2023-08-29:
Hello @PlayWeird,

Wuthouth the data we are basiaclly blind and there is not much we can say. The only recommendation would be to use any of the visualizers and see the points used for registration. Livox is typically sparse and sometimes there are almost no points left for the registration. For this I'd recommend by passing the double down sampling stattegy as already suggested in #128 and other issues as well.
The other parameter to keep an eye on it is the max range and the min range.

closed 2023-08-31 by nachovizzo

PlayWeird, 2023-08-31:
Thank you for the quick reply. I will try the things you suggest. I'm happy to provide you with .bag files if you are interested. Thanks again.