PRBonn/kiss-icp #216: How to run and postprocess KITTI dataset?

fghoussen, 2023-09-04:
How to run and postprocess KITTI dataset (ideally comparing with ground truth)?

I want to give a try to kiss-icp with KITTI dataset.

From here https://www.cvlibs.net/datasets/kitti/raw_data.php, one can download a sh script to download dataset:

```
>> cat raw_data_downloader.sh 
#!/bin/bash

files=(2011_09_26_calib.zip
2011_09_26_drive_0001)

for i in ${files[@]}; do
        if [ ${i:(-3)} != "zip" ]
        then
                shortname=$i'_sync.zip'
                fullname=$i'/'$i'_sync.zip'
        else
                shortname=$i
                fullname=$i
        fi
	echo "Downloading: "$shortname
        wget 'https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/'$fullname
        unzip -o $shortname
        rm $shortname
done

```

Running the sh script, I get:

```
>> tree -d 2011_09_26/
2011_09_26/
└── 2011_09_26_drive_0001_sync
    ├── image_00
    │   └── data
    ├── image_01
    │   └── data
    ├── image_02
    │   └── data
    ├── image_03
    │   └── data
    ├── oxts
    │   └── data
    └── velodyne_points
        └── data

```

As explained in the KITTI devlopment kit.

Now I try to use this dataset with kiss-icp

```
>> kiss_icp_pipeline  --dataloader kitti --sequence 01 --visualize 2011_09_26/
...
FileNotFoundError: [Errno 2] No such file or directory: '2011_09_26/sequences/01/calib.txt'

```

Is there some kind of transformation of the raw dataset to apply before using kiss-icp? If yes, how? Didn't see anything related to such a transformation in the doc. Not sure neither to understand what the sequence relates to: I set `--sequence 01` as I have `0001` in `2011_09_26_drive_0001 (in the example from `kiss_icp_pipeline --help` one see `kiss_icp_pipeline --dataloader kitti --sequence 07 --visualize <path-to-kitti-root>` but it seems there's no dataset labeled `0007` here https://www.cvlibs.net/datasets/kitti/raw_data.php?type=city whatever the category)

How can I run kiss-icp on KITTI dataset?

benemer, 2023-09-05:
Hey,

you are using the wrong dataloader. For KITTI Raw, you have to use the option `--dataloader kitti_raw`. The mapping between KITTI odometry sequence ID and the raw data folders can be found [here](https://github.com/PRBonn/kiss-icp/blob/5a10b2631d95201da241e75c29cfce68e3dbebec/python/kiss_icp/datasets/kitti_raw.py#L30):

```python
__raw_to_odometry_mapping__ = {
    0: "2011_10_03/2011_10_03_drive_0027_sync/",
    1: "2011_10_03/2011_10_03_drive_0042_sync/",
    2: "2011_10_03/2011_10_03_drive_0034_sync/",
    4: "2011_09_30/2011_09_30_drive_0016_sync/",
    5: "2011_09_30/2011_09_30_drive_0018_sync/",
    6: "2011_09_30/2011_09_30_drive_0020_sync/",
    7: "2011_09_30/2011_09_30_drive_0027_sync/",
    8: "2011_09_30/2011_09_30_drive_0028_sync/",
    9: "2011_09_30/2011_09_30_drive_0033_sync/",
    10: "2011_09_30/2011_09_30_drive_0034_sync/",
}
```

If you want to run it on other than these sequences, you could modify the dictionary above.

Best
Benedikt

closed 2023-09-05 by benemer

fghoussen, 2023-09-05:
@benemer: thanks!

As I'm new to the topic (lots of things are still blurry) my understanding of KITTI is that:
- KITTI odometry is motion-compensated: there (https://www.cvlibs.net/datasets/kitti/eval_odometry.php) you'll get these kind of datasets
[image]
- KITTI RAW is **not** motion-compensated: there (https://www.cvlibs.net/datasets/kitti/raw_data.php) you'll  get, for each dataset, 2 flavors of datasets like `2011_10_03_drive_0027_extract.zip` (raw data AFAIU) and/or `2011_10_03_drive_0027_sync.zip` (preprocessed data starting from raw data AFAIU).

After reading the KITTI website, I expected:
- KITTI RAW:
  - `extract` zip: raw data (so they should be called KITTI raw ?)
  - `sync` zip: preprocessed data (so they should be called KITTI ?)
  - difference between `extract` and `sync` zip:
    - images are rectified/undistorted
    - lidar points are ego-motion compensated
- KITTI odometry: subset of `sync` zip (as KITTI odometry = preprocessed motion-compensated data) from KITTI RAW according to a mapping (`00` == `2011_10_03_drive_0027`) explained in dev kit.

So far, so good. Now I discover `kiss-icp`...

If you could answer, some of these short questions, would be helpful!...
- Are my expectations above correct?
- If not:
  - What do you mean by "KITTI raw" (as there is 2 flavors of it - extract/sync) in `kiss-icp`? Seems `kiss-icp` uses "KITTI raw" to refer to (preprocessed) `_sync.zip` (https://nbviewer.org/github/nachovizzo/kiss-icp/blob/main/evaluation/kitti_raw.ipynb) although I expected refering to (raw) `_extract.zip`. Confused here.
  - If  `_sync.zip`  is KITTI raw data in `kiss-icp`, then when and why do you need `_extract.zip`? Does it make sense to use extract zip with `kiss-icp`? If no, why? Confused here too.
  - In both `2011_10_03_drive_0027_extract.zip` and `2011_10_03_drive_0027_sync.zip`  you'll find velodyne points: are they the same? Or are the first ones (extract zip) raw points, and the second ones (sync zip) ego-motion compensated points? And if so, which one to prefer using with `kiss-icp`? Can  `kiss-icp` handle both?
  - As sequence `00` is `2011_10_03/2011_10_03_drive_0027_sync/`, considering that `00` refers to "KITTI odometry motion-compensated" sequence, why does `kiss-icp` seems to consider a `sync` zip to be **raw** (= **not** compensated) data (following the doc here
[image]
where the KITTI RAW odometry link redirects to using `_sync` here https://nbviewer.org/github/nachovizzo/kiss-icp/blob/main/evaluation/kitti_raw.ipynb).

There is a lot of confusion in naming/words for newcomers and in crossing documentation (so much confusion that I hope I succeeded to make these questions clear). Hope you'll can answer these questions to shed light on this! KITTI has a lot of data: not always clear how to use them. Not always clear to get how to use them with `kiss-icp` neither...

benemer, 2023-09-06:
> Are my expectations above correct?

No, the KITTI Raw `sync` LiDAR scans are **not** motion compensated. There is an easy check to verify: Visualize a single point cloud **when the vehicle is in motion** and you will see that the scan lines are closed lines. However, since the vehicle moved, after one full rotation of the scanner the end point of a scan line will not lie next to the start point (since the vehicle moved). Therefore, the scans need to be motion compensated.

[image]

The two variants you refer to affect the synchronization and rectification of the other sensors (cameras, IMU). You can see that there are for example more IMU measurments than LiDAR scans in the `extract` folders. It should be:
- `extract `zip: unsynced+unrectified
- `sync` zip: synced+rectified

On the [webpage](https://www.cvlibs.net/datasets/kitti/raw_data.php) about KITTI Raw, the unsynced+unrectified data is called Raw and the synced+rectified is called processed. So, to be very precise, we would need to call `extract` the KITTI Raw Raw and the `sync` one KITTI Raw processed. Since most people always use the synced one, you usually refer to these when talking about KITTI Raw.

> And if so, which one to prefer using with kiss-icp? Can kiss-icp handle both?
The scans and timestamps are the same, the `sync` version just dropped some frames in the beginning and end to match it with the other sensor measurements. However, to use the `extract` data you will need to write your own dataloader because the points are stored in .txt files.

I hope I could help you.

fghoussen, 2023-09-10:
> we would need to call extract the KITTI Raw Raw and the sync one KITTI Raw processed

@benemer: thanks! It does help :)

benemer, 2023-09-19:
Our `kitti_raw` dataloader requires a sequence, but as I explained below, you can modify the dataloader and add your own sequence IDs. You can find the location of the file with

```python
python3 -c "import kiss_icp.datasets.kitti_raw as kitti_raw; print(kitti_raw.__file__)"
```

> Hey,
>
> you are using the wrong dataloader. For KITTI Raw, you have to use the option `--dataloader kitti_raw`. The mapping between KITTI odometry sequence ID and the raw data folders can be found [...]

fghoussen, 2023-09-19:
> you can modify the dataloader and add your own sequence IDs.

Indeed!