PRBonn/kiss-icp #221: What is correct_scan used for?

fghoussen, 2023-09-17:
What is correct_scan used for? Defined but not used AFAIU.

```
>> git log -1
commit f078513c92933d24b309e2853148b0ac06bbd414 (HEAD -> main, origin/main, origin/HEAD)
>> git grep correct_scan
python/kiss_icp/config/config.py:    correct_scan: bool = True
```

Is this meant only for KITTI like so (even if this will not work - not sure how to get `correct_scan` there)?

```
>> git diff
diff --git a/python/kiss_icp/datasets/kitti.py b/python/kiss_icp/datasets/kitti.py
index 197471d..cae7e77 100644
--- a/python/kiss_icp/datasets/kitti.py
+++ b/python/kiss_icp/datasets/kitti.py
@@ -65,7 +65,8 @@ class KITTIOdometryDataset:
     def read_point_cloud(self, scan_file: str):
         points = np.fromfile(scan_file, dtype=np.float32).reshape((-1, 4))[:, :3].astype(np.float64)
         #  points = points[points[:, 2] > -2.9]  # Remove the annoying reflections
-        points = self.correct_kitti_scan(points)
+        if correct_scan:
+            points = self.correct_kitti_scan(points)
         return points
 
     def load_poses(self, poses_file):
diff --git a/python/kiss_icp/datasets/kitti_raw.py b/python/kiss_icp/datasets/kitti_raw.py
index 2365882..bc14ff0 100644
--- a/python/kiss_icp/datasets/kitti_raw.py
+++ b/python/kiss_icp/datasets/kitti_raw.py
@@ -79,7 +79,8 @@ class KITTIRawDataset:
     def read_point_cloud(self, scan_file: str):
         points = np.fromfile(scan_file, dtype=np.float32).reshape((-1, 4))[:, :3].astype(np.float64)
         #  points = points[points[:, 2] > -2.9]  # Remove the annoying reflections
-        points = self.correct_kitti_scan(points)
+        if correct_scan:
+            points = self.correct_kitti_scan(points)
         return points, self.get_timestamps(points)
 
     def get_linear_velocity(self, idx):

```

nachovizzo, 2023-09-17:
The kitti dataset has a intrinsics calibration problem. That mean that the poincloud you read from the scans it's not so accurately projected to the 3D world. That correction step is just to minimize a bit this error, nothing else. With the proper raw data and the proper calibration this is not required :)

closed 2023-09-17 by nachovizzo

fghoussen, 2023-09-17:
So the parameter `correct_scan` in the config files has no effect? And will never have?

nachovizzo, 2023-09-17:
Good point (if you mean this one:)
https://github.com/PRBonn/kiss-icp/blob/5a10b2631d95201da241e75c29cfce68e3dbebec/python/kiss_icp/config/config.py#L30

Would you mind opening a PR to get rid of that one? So it's your contribution and not mine ;)

Otherwise let me know and I will do it. 

Thanks!