PRBonn/kiss-icp #226: Question - transformation convention \ understanding

ShaharSarShalom, 2023-09-21:
Hi 

with huge respect to this project, I susspect there be an algorithmic bug within the transformation

(File: kiss_icp.py, line: 59)

        initial_guess = last_pose @ prediction

last_pose represent transformation from frame 0 to N (am I right?)

predication represent transformation from frame N to frame N+1  (am I right?)

initial guess shall be transformation from frame 0 to N+1 if so, concatenating transformations shall be write on paper from "right to left", therefor I think that the correct form shoud be: 

        initial_guess = prediction @ last_pose

at this point I'd like to give huge appluse for this great project.

benemer, 2023-09-25:
No, `last_pose` transforms from frame N to frame 0 and `prediction` from N+1 to N. Therefore, `initial_guess` transforms from N+1 to 0.

There is an easy way to think about it: Imagine transforming the origin of the local frame N to frame 0. This should be the translation part of the pose at frame N (of `initial_guess`). If you write down the full transformation matrix `last_pose` and multiply with the origin `(0,0,0)` in homogeneous coordinates, you can see that `initial_guess` must transform from N to 0 to end up with the translation part.

closed 2023-09-25 by benemer

ShaharSarShalom, 2023-09-26:
Thank you so much for your explination

ShaharSarShalom, 2023-09-27:
Hi Benedikt 

I summeries the transformations within the code. I hope someone would find it usefull 

### The Kiss icp transformations explained 
1. ``pose`` - transform from frame N (current frame) to frame 0 
2. ``last_pose`` - transforms from frame N to frame 0 
3. ``prediction`` transforms from N+1 to N (transform one frame backward)

4. ``initial_guess = last_pose @ prediction``
initial_guess transforms from N+1 to 0.

5. ``new_pose`` - the ICP result between the current frame to the map. Transform from frame N+1 to 0. 

6. ``model_deviation = np.linalg.inv(initial_guess) @ new_pose``
represent how much the icp deviated from the initial_guess. which is actually the transformation between the hint to the icp result 

**At this point I'm a bit confused, I wanted to do another check,** 
I run the KITTI sequence 06. which is one of the simplest drives in KITTI. The car drives foward in a stright line for 1km. I run the first 50 frames and print the pose and prediction: 

prediction =
array([[ 9.99999839e-01, -5.44222752e-04, -1.58343716e-04, **1.24821898e+00**],
       [ 5.44272469e-04,  9.99999803e-01,  3.14106930e-04, 1.52556842e-02],
       [ 1.58172741e-04, -3.14193061e-04,  9.99999938e-01, 1.49858724e-02],
       [ 0.00000000e+00,  0.00000000e+00,  0.00000000e+00, 1.00000000e+00]])
  

last_pose = 
array([[ 9.99923493e-01, -5.12997140e-03,  1.12557307e-02, **6.00468358e+01**],
       [ 5.13077346e-03,  9.99986837e-01, -4.23826336e-05, 4.13453863e-01],
       [-1.12553651e-02,  1.00129995e-04,  9.99936651e-01, 6.65125245e-01],
       [ 0.00000000e+00,  0.00000000e+00,  0.00000000e+00, 1.00000000e+00]])

You can see the translation part in the predication and especially in the last_pose is positive, which actaully means that the transformation is from frame 0 to frame N. 

I look at it that way. lets say you are standing on frame 0, you add the translation part from last_pose - your  location would be close to frame N's frame place. that's how I find the pose transform from last_pose 0 to N (not the other way around). 

(Similarly that same for the predication transformation, if so concatenating the transformations is wrong)

I'd be very happy to hear from you

benemer, 2023-09-28:
I don't know which `pose` you are talking about. But if N+1 is the current frame (which we want to register), then `last_pose` transforms from N to 0, `prediction` from N+1 to N, and `initial_guess` and `new_pose` from N+1 to 0. Finally, `model_deviation` does not change the coordinate frame but measures the deviation within frame N+1.

> You can see the translation part in the predication and especially in the last_pose is positive, which actaully means that the transformation is from frame 0 to frame N.

> I look at it that way. lets say you are standing on frame 0, you add the translation part from last_pose - your location would be close to frame N's frame place. that's how I find the pose transform from last_pose 0 to N (not the other way around).

You are not correctly applying the transformation. You are correct that the car drives in positive x direction with respect to the LiDAR sensor frame, but this does **not** mean that the pose transforms from 0 to N. Let's say you want to know the location of the LiDAR at frame N expressed in the coordinate system of frame 0. This should be the translation part of the pose at frame N with positive value in x as you already said.

To do this, we take the location of the sensor at frame N expressed in frame N using homogeneous coordinates which results in `[0,0,0,1]`. Next, we use the pose at frame N which transforms **from frame N to frame 0**, right-multiply with the `[0,0,0,1]` vector, and get the expected positive translation in x.

If you were right and the pose transformed from 0 to N, you would need to invert it before multiplying with the sensor location expressed in frame N. Doing so, you would **not** get the expected result.

I hope this made it more clear.

ShaharSarShalom, 2023-09-28:
Thank you so much for your explination 
it's much more clear to me at this point