PRBonn/kiss-icp #233: Problem building ros2 wrapper colcon build

JuanCarlos-TiqueRangel, 2023-10-03:
Hello everyone I am trying to use the kiss-icp ros2 wrapper with ros galactic in a laptop with ubuntu 20.04. But after do the `git clone` I got the next problem in the colcon build:

```
error: no match for ‘operator=’ (operand types are ‘rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >::SharedPtr’ {aka ‘std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >’} and ‘std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >’)
   80 |         std::bind(&OdometryServer::RegisterFrame, this, std::placeholders::_1));
      |                                                                               ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/juan/AirSim/ros2/src/kiss-icp/ros/ros2/OdometryServer.cpp:24:
/usr/include/c++/9/bits/shared_ptr.h:309:19: note: candidate: ‘std::shared_ptr<_Tp>& std::shared_ptr<_Tp>::operator=(const std::shared_ptr<_Tp>&) [with _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’
  309 |       shared_ptr& operator=(const shared_ptr&) noexcept = default;
      |                   ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:309:29: note:   no known conversion for argument 1 from ‘std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >’ to ‘const std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >&’
  309 |       shared_ptr& operator=(const shared_ptr&) noexcept = default;
      |                             ^~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:313:2: note: candidate: ‘template<class _Yp> std::shared_ptr<_Tp>::_Assignable<const std::shared_ptr<_Yp>&> std::shared_ptr<_Tp>::operator=(const std::shared_ptr<_Yp>&) [with _Yp = _Yp; _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’
  313 |  operator=(const shared_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:313:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9/bits/shared_ptr.h: In substitution of ‘template<class _Tp> template<class _Arg> using _Assignable = typename std::enable_if<std::is_assignable<std::__shared_ptr<_Tp>&, _Arg>::value, std::shared_ptr<_Tp>&>::type [with _Arg = const std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >&; _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’:
/usr/include/c++/9/bits/shared_ptr.h:313:2:   required by substitution of ‘template<class _Yp> std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >::_Assignable<const std::shared_ptr<_Tp>&> std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >::operator=<_Yp>(const std::shared_ptr<_Tp>&) [with _Yp = rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > >]’
/home/juan/AirSim/ros2/src/kiss-icp/ros/ros2/OdometryServer.cpp:80:79:   required from here
/usr/include/c++/9/bits/shared_ptr.h:111:8: error: no type named ‘type’ in ‘struct std::enable_if<false, std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >&>’
  111 |  using _Assignable = typename enable_if<
      |        ^~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:324:2: note: candidate: ‘template<class _Yp> std::shared_ptr<_Tp>::_Assignable<std::auto_ptr<_Up> > std::shared_ptr<_Tp>::operator=(std::auto_ptr<_Up>&&) [with _Yp = _Yp; _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’
  324 |  operator=(auto_ptr<_Yp>&& __r)
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:324:2: note:   template argument deduction/substitution failed:
/home/juan/AirSim/ros2/src/kiss-icp/ros/ros2/OdometryServer.cpp:80:79: note:   ‘std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >’ is not derived from ‘std::auto_ptr<_Up>’
   80 |         std::bind(&OdometryServer::RegisterFrame, this, std::placeholders::_1));
      |                                                                               ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/juan/AirSim/ros2/src/kiss-icp/ros/ros2/OdometryServer.cpp:24:
/usr/include/c++/9/bits/shared_ptr.h:333:7: note: candidate: ‘std::shared_ptr<_Tp>& std::shared_ptr<_Tp>::operator=(std::shared_ptr<_Tp>&&) [with _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’
  333 |       operator=(shared_ptr&& __r) noexcept
      |       ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:333:30: note:   no known conversion for argument 1 from ‘std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >’ to ‘std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >&&’
  333 |       operator=(shared_ptr&& __r) noexcept
      |                 ~~~~~~~~~~~~~^~~
/usr/include/c++/9/bits/shared_ptr.h:341:2: note: candidate: ‘template<class _Yp> std::shared_ptr<_Tp>::_Assignable<std::shared_ptr<_Yp> > std::shared_ptr<_Tp>::operator=(std::shared_ptr<_Yp>&&) [with _Yp = _Yp; _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’
  341 |  operator=(shared_ptr<_Yp>&& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:341:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9/bits/shared_ptr.h: In substitution of ‘template<class _Tp> template<class _Arg> using _Assignable = typename std::enable_if<std::is_assignable<std::__shared_ptr<_Tp>&, _Arg>::value, std::shared_ptr<_Tp>&>::type [with _Arg = std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > > >; _Tp = rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >]’:
/usr/include/c++/9/bits/shared_ptr.h:341:2:   required by substitution of ‘template<class _Yp> std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >::_Assignable<std::shared_ptr<_Tp> > std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >::operator=<_Yp>(std::shared_ptr<_Tp>&&) [with _Yp = rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >, std::allocator<void> > >]’
/home/juan/AirSim/ros2/src/kiss-icp/ros/ros2/OdometryServer.cpp:80:79:   required from here
/usr/include/c++/9/bits/shared_ptr.h:111:8: error: no type named ‘type’ in ‘struct std::enable_if<false, std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::PointCloud2_<std::allocator<void> > > >&>’
  111 |  using _Assignable = typename enable_if<
      |        ^~~~~~~~~~~
[... cut ...]
make[2]: *** [CMakeFiles/odometry_component.dir/build.make:63: CMakeFiles/odometry_component.dir/ros2/OdometryServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:230: CMakeFiles/odometry_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< kiss_icp [4.59s, exited with code 2]

Summary: 3 packages finished [6.04s]
  1 package failed: kiss_icp
  1 package had stderr output: kiss_icp
```

To be honest I am not sure if is something with my ros version. Does anyone know what to do un this case ? Thank you in advance.

nachovizzo, 2023-10-03:
Hello, galactic end of life was last year in December 9th, 2022.... We are not planning on updating unsorted ROS distros

If you still want use galactic, you can give it a try by reverting #171, or by checking out before that change and post here if it worked out for other people that might be interested. 

But the best thing would be to update your unsupported ROS version ;) 

Good luck

closed 2023-10-03 by nachovizzo

cdebeunne, 2024-03-19:
Hello! I have managed to make it work on ROS galactic, I had the same issues as you. The problem is that `create_subscription<sensor_msgs::msg::PointCloud2>` doesn't support callbacks to `ConstSharedPtr`. I simply edited the callback to support `sensor_msgs::msg::PointCloud2` and edited the associated functions. Here are the changes I made:

- I edited the callback to support not const shared ptr msg in `OdometryServer.cpp`:
```c++
void OdometryServer::RegisterFrame(const sensor_msgs::msg::PointCloud2 &msg) {
    const auto cloud_frame_id = msg.header.frame_id;
    const auto points = PointCloud2ToEigen(msg);
    const auto timestamps = GetTimestamps(msg);
    const auto egocentric_estimation = (base_frame_.empty() || base_frame_ == cloud_frame_id);

    // Register frame, main entry point to KISS-ICP pipeline
    const auto &[frame, keypoints] = kiss_icp_->RegisterFrame(points, timestamps);

    // Compute the pose using KISS, ego-centric to the LiDAR
    const Sophus::SE3d kiss_pose = kiss_icp_->poses().back();

    // If necessary, transform the ego-centric pose to the specified base_link/base_footprint frame
    const auto pose = [&]() -> Sophus::SE3d {
        if (egocentric_estimation) return kiss_pose;
        const Sophus::SE3d cloud2base = LookupTransform(base_frame_, cloud_frame_id);
        return cloud2base * kiss_pose * cloud2base.inverse();
    }();

    // Spit the current estimated pose to ROS msgs
    PublishOdometry(pose, msg.header.stamp, cloud_frame_id);
    // Publishing this clouds is a bit costly, so do it only if we are debugging
    if (publish_debug_clouds_) {
        PublishClouds(frame, keypoints, msg.header.stamp, cloud_frame_id);
    }
}
```
- I edited the following functions of  `Utils.cpp`:
```c++
inline std::vector<Eigen::Vector3d> PointCloud2ToEigen(const PointCloud2 msg) {
    std::vector<Eigen::Vector3d> points;
    points.reserve(msg.height * msg.width);
    sensor_msgs::PointCloud2ConstIterator<float> msg_x(msg, "x");
    sensor_msgs::PointCloud2ConstIterator<float> msg_y(msg, "y");
    sensor_msgs::PointCloud2ConstIterator<float> msg_z(msg, "z");
    for (size_t i = 0; i < msg.height * msg.width; ++i, ++msg_x, ++msg_y, ++msg_z) {
        points.emplace_back(*msg_x, *msg_y, *msg_z);
    }
    return points;
}
...
inline std::vector<double> GetTimestamps(const PointCloud2 msg) {
    auto timestamp_field = GetTimestampField(msg);
    if (!timestamp_field.has_value()) return {};

    // Extract timestamps from cloud_msg
    std::vector<double> timestamps = ExtractTimestampsFromMsg(msg, timestamp_field.value());

    return timestamps;
}
...
inline std::optional<PointField> GetTimestampField(const PointCloud2 msg) {
    PointField timestamp_field;
    for (const auto &field : msg.fields) {
        if ((field.name == "t" || field.name == "timestamp" || field.name == "time")) {
            timestamp_field = field;
        }
    }
    if (timestamp_field.count) return timestamp_field;
    RCLCPP_WARN_ONCE(rclcpp::get_logger("odometry_node"),
                     "Field 't', 'timestamp', or 'time'  does not exist. "
                     "Disabling scan deskewing");
    return {};
}
...
inline auto ExtractTimestampsFromMsg(const PointCloud2 msg,
                                     const PointField &timestamp_field) {
    auto extract_timestamps =
        [&msg]<typename T>(sensor_msgs::PointCloud2ConstIterator<T> &&it) -> std::vector<double> {
        const size_t n_points = msg.height * msg.width;
        std::vector<double> timestamps;
        timestamps.reserve(n_points);
        for (size_t i = 0; i < n_points; ++i, ++it) {
            timestamps.emplace_back(static_cast<double>(*it));
        }
        return NormalizeTimestamps(timestamps);
    };

    // According to the type of the timestamp == type, return a PointCloud2ConstIterator<type>
    using sensor_msgs::PointCloud2ConstIterator;
    if (timestamp_field.datatype == PointField::UINT32) {
        return extract_timestamps(PointCloud2ConstIterator<uint32_t>(msg, timestamp_field.name));
    } else if (timestamp_field.datatype == PointField::FLOAT32) {
        return extract_timestamps(PointCloud2ConstIterator<float>(msg, timestamp_field.name));
    } else if (timestamp_field.datatype == PointField::FLOAT64) {
        return extract_timestamps(PointCloud2ConstIterator<double>(msg, timestamp_field.name));
    }

    // timestamp type not supported, please open an issue :)
    throw std::runtime_error("timestamp field type not supported");
}

```
I hope this will help.

JuanCarlos-TiqueRangel, 2024-04-04:
@cdebeunne Thank you very much for your help. I follow your instructions and works pretty well, thank you!