PRBonn/kiss-icp #237: reset initial pose through ros service

RamanaBotta, 2023-10-10:
Hello, 
### About
Since kiss-icp is prone to accumulation of positional errors over time, having something like a service call to reset odometry would really help when working with other sensors, especially sensors like GPS (globally consistent).
### Existing solutions
This kind of feature is available with Zed Cemera positional tracking, Where through a service call it we can reset the start position of odometry [Ref](http://wiki.ros.org/zed-ros-wrapper).

nachovizzo, 2023-10-10:
I agree this would be great to have. Would you like to contribute it to the project? I can guide you a bit once you have a prototype ready ;)

RamanaBotta, 2023-10-12:
I am happy to contribute :)

RamanaBotta, 2023-10-12:
I found a related code snippet in the Zed wrapper.
https://github.com/stereolabs/zed-ros-wrapper/blob/71eb2bb434f059e17191503b707267938f5a1b7f/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp#L408

RamanaBotta, 2023-10-12:
Hello, please find the pull request regarding this issue.