PRBonn/kiss-icp #239: Pose drift with large hesai xt32 rosbag

JzHuai0108, 2023-10-12:
Dear authors,

Thanks for the great work.
We tried the kiss-icp with Hesai XT32 rosbags captured by a handheld device. It worked pretty well for a small loop in a basketball court with the perimeter around 40 meters. 
[image]

Then we tried it with a >100 GB rosbag. At the beginning, it also worked seemingly good. But over time, the trajectory became more erratic. 
[image]
We used the below command
```
roslaunch kiss_icp odometry.launch
```
The key parameters 
```
  <arg name="deskew" default="true"/>
  <arg name="max_range" default="120.0" />
  <arg name="min_range" default="2.0" />
  <arg name="voxel_size" default="1" />
```
How can we make it work better with the large bag file?

nachovizzo, 2023-10-12:
Hello, thanks for the detailed information. It's always best to share the data to try it ourselves.

This is the typical problem: we are downsampling the input cloud too much. 

## Python Fix
You can try much more easily the python interface (`pip install kiss-icp`) and then apply this fix: https://github.com/PRBonn/kiss-icp/issues/128#issuecomment-1505258331

## ROS / C++ Fix

For ROS (you don't specify, but I'm assuming you are running ROS 1), you can hack a bit the pipeline and recompile by doing:
```diff
diff --git a/cpp/kiss_icp/pipeline/KissICP.cpp b/cpp/kiss_icp/pipeline/KissICP.cpp
index 3e06f7f..ba6094b 100644
--- a/cpp/kiss_icp/pipeline/KissICP.cpp
+++ b/cpp/kiss_icp/pipeline/KissICP.cpp
@@ -83,8 +83,7 @@ KissICP::Vector3dVectorTuple KissICP::RegisterFrame(const std::vector<Eigen::Vec
 KissICP::Vector3dVectorTuple KissICP::Voxelize(const std::vector<Eigen::Vector3d> &frame) const {
     const auto voxel_size = config_.voxel_size;
     const auto frame_downsample = kiss_icp::VoxelDownsample(frame, voxel_size * 0.5);
-    const auto source = kiss_icp::VoxelDownsample(frame_downsample, voxel_size * 1.5);
-    return {source, frame_downsample};
+    return {frame_downsample, frame_downsample};
 }
 
 double KissICP::GetAdaptiveThreshold() {
```

## Related issues

- https://github.com/PRBonn/kiss-icp/issues/128
- https://github.com/PRBonn/kiss-icp/issues/146
- https://github.com/PRBonn/kiss-icp/issues/215
- https://github.com/PRBonn/kiss-icp/issues/186

JzHuai0108, 2023-10-13:
1. I tried less harsh downsampling, but the result looked bad on the small sequence.
2. The small sequence and large sequence are shared [here](https://1drv.ms/f/s!ArHbZ5OvjCBgh2PmTu8gXhDNtjgD?e=mOAWAq).
The small one is about 4 Gb, the large one is about 66 Gb. The data were collected by Hesai XT32. The GNSS/INS solution is also available, but has jumps due to GNSS outages caused by big trees lined on the roadside.
It was raining, so the camera images were not clear.
Note the bustling campus may cause problems to ICP.

benemer, 2023-10-13:
When inspecting your data, I realized a weird pattern, see the attached image. It looks like there are two overlapping lidar scans, also because the scans have points below the ground surface. Do you have an idea why?

[image]

JzHuai0108, 2023-10-13:
The device was held by a walking person. You may see the zed images in the rosbag.
There was moderate to light raining during the data collection. 
The flat basketball court covered by plastic rubber was pretty wet. This may explain some points underground? 
The big sequence is still being uploaded :).

benemer, 2023-10-16:
Alright, I think this explains the reflections! And also the shakiness of the data since it is a walking person.

I am currently downloading the larger dataset. Will get back to you soon :)

benemer, 2023-10-18:
There are multiple reasons KISS is having a hard time in this dataset. 

First, the LiDAR is rather sparse with 32 beams and a limited field of view (half of the scan is missing due to occlusion). 

Second, the motion profile is quite shaky since it is a handheld. 

Third, the rain imposes an issue because the beams get reflected at the raindrops which look like random points in the scan plus the reflections on the ground that depend on the viewing angle.

One thing I noticed with 32-beam sensors (for example, the one used in Nuscenes) is that limiting the maximum range improves performance. So with this config:

```yaml
data:
  deskew: True
  max_range: 50.0 # can be also changed in the CLI
  min_range: 2.0
```

I managed to run KISS successfully for the first 3,000 frames, so that might be a direction to follow!

[image]

JzHuai0108, 2023-10-19:
Thank you for your time and patience.
Your points are convincing.
One point I'd like to add is that the 120 m range lidar's beam is more likely to come across rain drops in the round trip as the distance increases. This may also support the decision to use a shorter range than the spec value.
Do you have a paypal or buy me a coffee or alipay code? I'd like to buy you a beer.

benemer, 2023-10-19:
That's true! However, the points we mask out using `max_range` are those that did not get reflected. If there is a reflection within `max_range`, there is not much we can do and we keep the point.

I believe the sparsity issue becomes more severe for far-away points which makes it harder to align the scan to the sparse local map, especially when facing changing pitch angles as in your dataset.

Here is the result of the full run using the config from above.

[image]

> Do you have a paypal or buy me a coffee or alipay code? I'd like to buy you a beer.

All good, but thanks a lot! Maybe there will be an occasion to have a drink when meeting at a conference in the future :)

Best
Benedikt

JzHuai0108, 2023-10-20:
Thanks a lot for your time. We will run the KISS-ICP throughout our datasets. I will close the issue for now.

> Maybe there will be an occasion to have a drink when meeting at a conference in the future :)

Definitely.

closed 2023-10-20 by JzHuai0108

benemer, 2023-11-06:
Hey,

are you talking about the data you shared with me? Or is this another run?

As mentioned above, I kept the original downsampling scheme and used the config mentioned [above](https://github.com/PRBonn/kiss-icp/issues/239#issuecomment-1768704956). Please check again.

Best
Benedikt

JzHuai0108, 2023-11-07:
Thank you for the quick reply.
This is the same data I shared with you. 
OK. I will try the program with your setting on another computer.