PRBonn/kiss-icp #242: How would I go about making this work for my dataset?

themightyoarfish, 2023-10-13:
I have a dataset that is captured from above, basically like a drone where the LIDAR is looking down. The device has 128 layers, so I would have expected the registration to be successful, but the pipeline does not at all detect any lidar motion and the resulting map is thus entirely smeared.

The scenario is this and the lidar moves perpendicularly to its shorter FOV

[image]

Some stationary points and some noise via rain (i think) is present, but other than that I don't see any obvious confounders.

When I run this with `kiss_icp_pipeline --visualize`, I get nothing useful as the scanner starts moving:

Is there anything obvious here that would make it fail?

[kiss_icp.webm](https://github.com/PRBonn/kiss-icp/assets/11613312/961c23f7-e421-47b8-b70d-72568e81f0c6)

I've uploaded the data here if that helps: https://drive.google.com/file/d/1UhD59qZhV7QWhZXtJjHH9GSOM4cLobsZ/view?usp=sharing

nachovizzo, 2023-10-16:
@themightyoarfish Thanks for the info and the data.

Honestly, KISS-ICP would not work on that dataset. As an experiment, I played back the dataset with `lidar_visualizer` (just `pip install lidar-visualizer` and then `lidar_visualizer <folder>`), and honestly, I couldn't figure out what would be the expected motion. I guess that you will need to to other sources of information like wheel odometry, IMU's etc

Good luck

closed 2023-10-16 by nachovizzo

themightyoarfish, 2023-10-16:
If you know the scenery it's relatively obvious what direction the scanner is moving, and the data density is high, so adjacent scans should be rather close. Therefore my question is what specifically makes this harder for KISS-ICP? Is there some implicit assumption that the motion is more or less in the same plane as the LIDAR's horizon, like in a car for instance?