PRBonn/kiss-icp #254: Tilted lidar

arixrobotics, 2023-11-09:
Hi, 

I'm using a tilted down lidar (VLP16) on a mobile base. How should I tell kiss-icp my lidar is tilted?

nachovizzo, 2023-11-09:
🤔 You don't. The system is independent from where the sensor is mounted. If you want to transform the poses to another coordinate frame, then you will need the calibration. But is hard to say without the full picture

PS: If you are using VLP16 please check #240

closed 2023-11-09 by nachovizzo

arixrobotics, 2023-11-09:
Kiss-ICP ftw!
Turns out I could just do with some parameter tunings, no need to make it less downsampled. 

With the default parameters, I get a trajectory with wrong elevation:
[image]

With `$ roslaunch kiss_icp odometry.launch topic:=/velodyne_points max_range:=20 voxel_size:=0.5 min_range:=0.0` it manages to register correctly, showing the lidar going through the same path to and fro correctly:
[image]

Thanks a lot!