PRBonn/kiss-icp #255: Building error for ROS2

SangHunHan92, 2023-11-09:
Hello, I'm using ROS2 Foxy, Ubuntu 20.04, and I am getting error when building package.

I also show this repo(https://github.com/PRBonn/kiss-icp/issues/76), I still have same error.

[error_log.txt](https://github.com/PRBonn/kiss-icp/files/13307053/error_log.txt)

nachovizzo, 2023-11-09:
Hello, thanks for reporting. Is the expected behavior we **do not support** foxy since https://github.com/PRBonn/kiss-icp/pull/171. It has reached end-of-life and therefore it does not have official support anyways. Please move to `humble` or `iron`, they are much better :)

Commit where we dropped support (in case you want to checkout and use an old version of KISS-ICP) : [1e0c28e](https://github.com/PRBonn/kiss-icp/pull/171/commits/1e0c28e6e11fb45aba6de507ab6c997e36f991be)

closed 2023-11-09 by nachovizzo

SangHunHan92, 2023-11-15:
I changed to humble, and it's working well. Thanks !