PRBonn/kiss-icp #257: Not working well in some situation

SangHunHan92, 2023-11-14:
Hi, thank for the ease of use and nice result !

However, I'm getting bad results in some situations.
I download dataset from here(https://github.com/TixiaoShan/LIO-SAM), and run kiss-icp with garden_dataset.bag and rotation_dataset.bag.

[image]

This looks like an incorrect pose was predicted during the registration process.

Can you tell me when these situations will appear? 
Also, can you tell me about parameter tuning or hardware settings to overcome this?

benemer, 2023-11-17:
Hey,

I had a brief look and can confirm what you observed. My first guess is that the motion is too shaky. For example, in the `rotation_dataset.bag` the sensor starts rotating fast after the first few frames and our local map will not be a good target for registration yet. In general, our constant velocity assumption will no longer hold for abrupt changes of direction which you can observe in both datasets.

In such cases, you will probably need to use information from other sensors, like the IMU.

Best
Benedikt

SangHunHan92, 2023-11-17:
Thank very much. I now know what I need to do to capture the dataset.

closed 2023-11-17 by SangHunHan92