PRBonn/kiss-icp #259: Robustness under frame rate changes

zyxu1996, 2023-11-23:
I used a 30-second 10HZ point cloud and randomly dropped 20 frames of it. I noticed that pose's estimates were not ideal, and the point clouds did not match up well on discrete times. Is there any way to adjust the parameters in config to deal with this? Or do I have to use an IMU to optimize it?

nachovizzo, 2023-11-23:
Would you mind explaining the meaning of:

> I used a 30-second 10HZ point cloud and randomly dropped 20 frames of it.

zyxu1996, 2023-11-24:
There's a point cloud with a timestamp like this

zyxu1996, 2023-11-24:
There's a discontinuous point cloud with a timestamp like this (s means second):
0.1s, 0.2s, 0.3s, 0.5s, 0.7s, 0.8s, 0.9s, 1.0s, 1.4s, 1.5s, 1.6s
It lost 0.4s, 0.6s, 1.1s, 1.2s, 1.3s. It will most likely give a wrong pose prediction at 1.0s or 1.4s.

zyxu1996, 2023-11-24:
> Would you mind explaining the meaning of:
> 
> > I used a 30-second 10HZ point cloud and randomly dropped 20 frames of it.

There's a discontinuous point cloud with a timestamp like this (s means second):
0.1s, 0.2s, 0.3s, 0.5s, 0.7s, 0.8s, 0.9s, 1.0s, 1.4s, 1.5s, 1.6s
It lost 0.4s, 0.6s, 1.1s, 1.2s, 1.3s. It will most likely give a wrong pose prediction at 1.0s or 1.4s.

benemer, 2023-12-04:
Our [prediction](https://github.com/PRBonn/kiss-icp/blob/3465687027471419acb1e425d2e22ca02a360925/python/kiss_icp/kiss_icp.py#L87) model

```python
    def get_prediction_model(self):
        if len(self.poses) < 2:
            return np.eye(4)
        return np.linalg.inv(self.poses[-2]) @ self.poses[-1]
```
assumes that the scans are equidistant in time (therefore we can use the delta between the last two estimated poses as the initial guess for the next one). This does not hold in your case, so you will sometimes use the pose change within 0.2 seconds as an initial guess for a movement within 0.1 seconds (which is wrong).

The first thing you should try is taking the actual time difference between estimations into account to predict an initial guess.

tizianoGuadagnino, 2023-12-07:
@zyxu1996 KISS is not meant to handle specific hardware issues such as frame dropping. This probably require some robustification of the pipeline to wrong initial guesses, which is not the goal of KISS.  As such, if this is needed for your research, I suggest you fork the code and work on it. If you find anything interesting, please share it with our community by reopening this issue.

closed 2023-12-07 by tizianoGuadagnino