PRBonn/kiss-icp #261: Map not dense in uneven terrain

stapperfend, 2023-12-13:
We collected some LIDAR data (Ouster OS0-128) with a small and slow moving mobile robot on a volcanic area with an uneven surface. We now want to create a map using the ros2 bag files. It seems like KISS-ICP is having issues with the data, as the created local map is very coarse and no real dense map is created. 
[image]

We tried to play around with the config parameters, but it is also not clear to us, where we need to make the changes, as there are several locations where the config may be edited:
- /ros/launch/odometry.launch
- /config/basic.yaml
- /python/kiss_icp/config/config.py
We adjusted some parameters in all those files, but it looks like it changed nothing. If I am not mistaken,  the `min_range `and `max_range` values should at least change where the keypoints appear. I adjusted `min_range` to 0.0 and `max_range` to 3.0. But as it appears in the following image, the keypoints are only detected further away:
[image]

Could you please let me know where to adjust parameters for the usage of KISS-ICP in ROS 2 and if possible give a quick look at our ros2 bag?

The ros2 bag is available here:
https://drive.google.com/file/d/1FlzOM4woJm3kN_WOAuYiy9_HIkZ7u3wc/view?usp=sharing

nachovizzo, 2023-12-18:
@stapperfend thanks for reporting, I'm with a spotty connection so I will only be able to look at the data in a few weeks. You will likely hear from us next year ;)

benemer, 2024-01-22:
Hey!

> as the created local map is very coarse

This is intended. Please note that the local map of KISS is used as a target for registration and this map is downsampled to reduce runtime. I will cite @nachovizzo from a [previous](https://github.com/PRBonn/kiss-icp/issues/14#issuecomment-1278817308) issue:

> Our internal map representation is only used to obtain the incremental pose estimation and by no means should be confused with a robotics map. It's just a local representation of the environment, and it is not a global mapping solution.

So please have a look at some existing issues covering this topic like https://github.com/PRBonn/kiss-icp/issues/14, https://github.com/PRBonn/kiss-icp/issues/90, and https://github.com/PRBonn/kiss-icp/issues/223.

In short, you can use the poses estimated with KISS together with for example [VDBFusion](https://github.com/PRBonn/vdbfusion) to get an actual map.

tizianoGuadagnino, 2024-01-29:
Hi @stapperfend,

Did you try to change the parameters in the [ROS2 launch file](https://github.com/PRBonn/kiss-icp/blob/main/ros/launch/odometry.launch.py) ? I think you were modifying the wrong ROS launch.

Let us know if any of this help.

tizianoGuadagnino, 2024-03-01:
I close this due to inactivity, if need it you can open it again and I am happy to help you.

closed 2024-03-01 by tizianoGuadagnino