PRBonn/kiss-icp #264: Bad results using a dome sensor (32 scan lines 90° x 360°)

hofbauerc, 2024-01-08:
Recently we received a new sensor which we would like to use for odometry purposes. The sensor for this case is a Robosense BPearl. The reason for this sensor was the large FoV with 90° x 360° (hemispherical sensor). We normally use an ouster OS1 64 for our odometry which worked really well in the past. However, we have some issues with the new sensor. 

As mentioned in this issue https://github.com/PRBonn/kiss-icp/issues/128#issuecomment-1505258331 , the reason could be the sparse pointcloud, which will be filtered due to the development of Kiss-ICP for larger clouds. Using the fix in this issue however, it does not seem to affect the results of the odometry that much. 

Are there any other possible fixes for this problem? I could also provide a ros1 or ros2 bag if it helps.

Here a video of the problem:

https://github.com/PRBonn/kiss-icp/assets/116653994/9eb46dda-6da3-4766-8fbd-75a0dc99ddb6

The map should look something like this at the start (captured with ouster OS1 64):
[image]

But with the robosense (including the fix from https://github.com/PRBonn/kiss-icp/issues/128#issuecomment-1505258331) it looks like this:
[image]

hofbauerc, 2024-01-23:
Hi,

Thanks for the help. Maybe you manage to tune the algorithm to work with this specific (rather "sparse") pointcloud.

Here is the link to the bag, I hope it works: https://drive.google.com/file/d/11a6VCGWfTcjk541i-dv1IZGV31KJ95rX/view?usp=sharing

The Bag also includes ouster data for reference to see the difference between the two sensor systems.

Thanks in advance!

benemer, 2024-01-25:
Thanks! I had a look at the data and ran a few experiments. The main issue is the sparsity. As soon as the platform is in motion, KISS tries to align the scanlines to the previously added scanlines in the map. To mitigate this effect, I reduced the max range (the farther away the points, the more sparse they are) and increased the voxel size, breaking the map's scanline pattern.

Note that this is just based on guessing and it is not a "general recipe" for sparse sensors. But at least for your data it works :)

It would be nice to investigate this a bit further in the future.

[image]

```yaml
data:
  max_range: 50.0
  min_range: 0.0

mapping:
  voxel_size: 2.0
```

hofbauerc, 2024-01-26:
Thank you very much. I already tried the limit to 50m (and lower) but did not change the voxel size. I will check those parameters on some other data that I recorded. Thanks again for the help.

closed 2024-01-29 by tizianoGuadagnino