PRBonn/kiss-icp #266: An ICP mismatch problem

JzHuai0108, 2024-01-14:
I think the idea of using simple ICP to match point clouds brilliant, so I try to apply this idea to sparse radar point clouds.
We collected a sequence of radar point clouds from a basketball court.
With an established method based on fast-lio2, we achieve the below mapping result.
[image]
It is impressive because it does not drift over many loops. This is the effect I try to attain, no drift in a small area.

However, with the ICP I implemented, it always drifts. see the below picture.
[image]
The ICP just does not correct the drift at turns, leading to multiple lines corresponding to the wall. 
I tried all sorts of ways to tune the implementation. Nothing seems to improve the result.
The parameters I tuned include the voxel size, the number of points in a voxel, the matching point distance threshold, etc.

Spending more than 1 month on this problem, I rewrite the implementation twice, the results just got worse.
I have perused the kiss-icp ra-l paper, fast-lio2 t-ro paper several times, and I have discussed with the iriom primary author multiple times, but unable to find any clue.
I know you guys are very experienced in ICP matching, can you please give me some light to get out of the pitfall?
My current working direction is to adjust the weights of ICP matches according to the planeness of a query point's neighbors.

JaySlamer, 2024-01-22:
Came across a similar problem when working with my USV dataset. when the boat is cycling around a bridge, the result is terrible.

benemer, 2024-01-22:
Can you please share the data? Otherwise, it is hard to say what is going on. Thank you!

JzHuai0108, 2024-01-24:
Hi Benedikt,

The rosbag (ros1 version) is available at [here](https://1drv.ms/f/s!ArHbZ5OvjCBgh3w-NR2uetB-cWGB?e=TI1Gk2).
It may take up to 2 hours from now to finish the uploading.
The extrinsics.md has some basic info about sequence.
The odometry of interest to me is 4d radar inertial odometry.
If you think this is beyond your scope, I totally understand.

Regards,

benemer, 2024-01-25:
Thanks! That's indeed some challenging data due to the radar sensor's noise and the sparsity. I just tried different configurations, but I agree with you that integrating the IMU is the way to go.

Maybe @tizianoGuadagnino can comment on this, he recently worked together with our colleague Yibin Wu on this topic. [Here](https://arxiv.org/pdf/2311.09887.pdf) is a link to their arXiv paper.

Best
Benedikt

JzHuai0108, 2024-01-26:
LIO-EKF sounds like the extension to KISS-ICP with IMU data. I think it may work with sparse and noisy 4d radar point clouds.
To make the odometry estimation more reliable, I think it is necessary to use its doppler measurements.

tizianoGuadagnino, 2024-01-29:
Hi @JzHuai0108,

I think LIO-EKF will be more suitable for your scenario. The IMU's help is essential when you work with radar data. The only problem is that, in LIO-EKF, we use a single iteration of the EKF-style ICP. That shouldn't be sufficient with noisy radar data. I am confident that, with multiple iterations, you will obtain a result similar to fast-lio2.

 I also agree that exploring the Doppler measurements might be helpful in this case. I'm not a radar person, so I wonder how difficult it is to model in an EKF-like scheme. Do you have any ideas already on how to do that? I think also @YibinWu might be interested also.

JzHuai0108, 2024-01-30:
We are working on this, integrating doppler measurements and scan to submap matches in an iterative refinement scheme. It is actually pretty straightforward to integrate the doppler measurements, once you have the formula relating the ego velocity and the doppler measurement. I will come back if I get some decent result.

tizianoGuadagnino, 2024-01-30:
Awesome, let us know what you find out :)

closed 2024-01-30 by tizianoGuadagnino