PRBonn/kiss-icp #271: Parameter tuning in the config file

Toytiny, 2024-01-28:
Hi,

We are trying to use kiss-icp to do odometry for rgbd point cloud captured in the helmet camera. The head hardly moves in the sequence, with a maximum of millimetres in several frames. Could you suggest any parameters combinations for the adaptive_threshold:
  initial_threshold: 
  min_motion_th:

tizianoGuadagnino, 2024-01-29:
Hi @Toytiny,

The issues on this repo are meant for the community to help us improve KISS-ICP and make it more usable. We are not here to tune the parameters for your specific scenario unless there is a specific reason for it. If you highlight a specific scenario that is represented by your data and can be useful to other people we are happy to help. Anyway, without seeing the data it is impossible to help you.

tizianoGuadagnino, 2024-01-29:
If you just tried the system and it didn't work, it is better if you tune it a little bit by yourself and then open an issue.