PRBonn/kiss-icp #274: Missing file "setup.py" when calling $make editable

ErnestF22, 2024-02-06:
PIP Editable installation following instructions provided in the [Python README file](https://github.com/PRBonn/kiss-icp/blob/main/python/README.md)  fail, stating that there is a missing "setup.py" file (screenshot attached).

[image]

nachovizzo, 2024-02-06:
@ErnestF22 would you mind sharing a bit more details of your system, `pip` version, etc?

`editable` mode is a big *pain*, and it depends on which version of `pip` you have installed the behavior you would get. Usually `pip install --upgrade pip` would give you the latest pip version and is very likely to work 

In my case `24.0` works fine

```
pip 24.0 from /home/ivizzo/.venv/lib/python3.10/site-packages/pip (python 3.10)
```

ErnestF22, 2024-02-06:
Hi @nachovizzo and thanks for the quick answer. As you suggested, I had an older version of pip installed.
However, only upgrading it did not really fix anything.
Thankfully, after upgrading pip I went and did a full reinstallation of KISS-ICP (using `$ pip install "kiss-icp[all]"`) and then for some reason `$make editable` finally worked. Maybe the pip upgrade changed the minimum required versions of some packages that were causing the error, and then everything got fixed from that.

From my side, this issue can be closed.
Thanks once again.

closed 2024-02-06 by nachovizzo