PRBonn/kiss-icp #277: How to evaluate modified code?

leedaehan, 2024-02-21:
Hi, First of all, thank you for your great work!

I modified the `GetCorrespondences` function in kiss-icp's **cpp** file and want to evaluate it on the newer college dataset. 

So I would like to ask what steps should be taken to evaluate modified code using `newer_college.ipynb`.

Thank you in advance!

nachovizzo, 2024-02-22:
Open the notebook and hit run :)

closed 2024-02-22 by nachovizzo

leedaehan, 2024-02-22:
I already tried it that way, but it seemed like the original KISS-ICP was executed without the modified code being reflected.

To integrate the changes, I followed the pybind11 build process outlined below, executed within the kiss_icp/python/ directory:
`mkdir build && cd build`
`cmake ..`
`make install`

However, I'm encountering difficulties in applying the modified code to evaluate the relative translational error in percentage with the newer college dataset.

After building `kiss_icp_pybind.cpp`, What are the necessary steps to successfully apply the modified code?