PRBonn/kiss-icp #297: using kiss-icp yields only tranformations existing out of 0's and 1's

joostzaalberg, 2024-03-20:
Hello reader, 

For a school project, I'm currently trying to build a map of the static environment of a nuScenes scene. First, I remove al dynamic and non-dynamic cars, pedestrians, etc., at this point, I only have lidar frames of the static environment. Using only the geo location of the data set, the alignment of these frames is too poor for further processing. Here is where I want to use kiss-icp to align the different frames. I exported the frames into .obj files (in x, y, z) and placed them in a folder. The frames are already pretty close to each other (few cm's). When I run the kiss_icp_pipeline on the folder with all the 381 frames, it only yields a tum file where the first column counts from 1 to 381, the remaining but the last columns are zero, and the last column is 1. (the Kitthy file yields only ones and zeros too) What could this be? I tried to randomly translate the frames by a few meters, to make it clear that the frames are not aligned, but the same result. Any suggestions? Thanks for the help!! Best, Joost

joostzaalberg, 2024-03-20:
[latest.zip](https://github.com/PRBonn/kiss-icp/files/14668162/latest.zip)

Here are my result files in a .zip, if it helps. Is there any way I can create more elaborate log files I can share?

nachovizzo, 2024-03-20:
Have you tried using the nuScenes dataloader we provide with kiss?

closed 2024-03-22 by nachovizzo

joostzaalberg, 2024-03-25:
Hi, thanks for your reply. Unfortunately, it has not been resolved yet. I have not used it because I do not use an entire scene. As mentioned, while exporting the point cloud as a .obj file, I remove dynamic categories such as cars and pedestrians. However, using these .obj files kiss does not yield results. Any other suggestions?

joostzaalberg, 2024-03-25:
[sample000 copy.txt](https://github.com/PRBonn/kiss-icp/files/14746340/sample000.copy.txt)

If it helps any, this is how one of the .obj files looks like (I converted it to txt so I could upload it to GitHub)

nachovizzo, 2024-03-25:
we cant tell by looking to a txt file. It looks like you need to get your data right, probably your clouds are in a completely wrong coordingate frame. By just inspecting the indices, seems like it's on global frame instead of lidar frame.

joostzaalberg, 2024-03-30:
I got it to work now that my points are in the sensor frame, thanks! It does work a lot better then the original ego motion data of nuscenes, :)