PRBonn/kiss-icp #299: Deskew model

kulfer80, 2024-03-24:
Hi, first of all, thank you for your great work!

I have a consideration about de-skew model. I'm not that comfortable in concepts of lie algebra, but I don't totally understand, why do we use parameter `mid_pose_timestamp` ? I have read paper but I didn't find clear explanation. For me the thing that is logical that first point should be just `corrected_frame[0] = motion * frame[0];`, where motion in this case should have been identity transformation matrix. But when using `mid_pose_timestamp` it will yield to different result isn't it ? You are all for sure better experts in this field, but I just wanted to ask this stupid question to understand broader context about this beautiful and well written masterpiece.

https://github.com/PRBonn/kiss-icp/blob/994d2329effaec481b88adf2a0d2aa1a9ec40187/cpp/kiss_icp/core/Deskew.cpp#L44

Thank you for your time and effort.

nachovizzo, 2024-03-25:
@kulfer80 good question, I have been waiting months until someone asks this.

I'm currently a bit overloaded but I promise I will come back with a detailed explanation. In the meantime, you can rest assured, that 0.5 makes total sense

changh95, 2024-03-26:
TLDR; Based on my experience, it's critical to use mid_pose_timestamp for effective KISS-ICP operation.

I'm working on a skewed LiDAR dataset, where each LiDAR frame's timestamp was set at the start of its scanning sequence. I used `const auto motion = Sophus::SE3d::exp(timestamps[i] * delta_pose)` to deskew the LiDAR frames, based on the premise that the LiDAR's movement could be approximated by a constant velocity model. This approach seemed reasonable, as it intended to correct distortions more significantly at the end of the scanning sequence than at the beginning.

However, through my experience, I realized a significant pitfall: the constant velocity model is not flawless and introduces errors that get woven into the LiDAR points during the deskewing process. When applied with `const auto motion = Sophus::SE3d::exp(timestamps[i] * delta_pose)`, these errors become proportionally tied to the timestamps, significantly affecting the later points in the sequence. This is particularly problematic for KISS-ICP, which depends on precise point-to-point comparisons, and these errors can escalate, particularly when compounded over several frames.

From my experience, I've found that deskewing LiDAR frames to the midpoint of the scanning sequence is significantly more effective in reducing errors compared to deskewing to the beginning of the sequence. This understanding is based on the concept that deskewing errors are directly related to the distance of the timestamp from the reference point used in the deskewing process. By setting the reference point to the midpoint (0.5) of the scanning sequence, the errors are minimized, which is crucial for the reliable operation of the KISS-ICP algorithm. The accompanying chart illustrates this point, with yellow representing errors when deskewing to the midpoint and red indicating errors when deskewing to the start of the sequence. This visualization underscores the advantage of using a midpoint reference, as it leads to the lowest deskewing errors, enhancing the reliability of KISS-ICP in processing the LiDAR data.

[image]

kulfer80, 2024-04-01:
Thank you very much for the detailed explanation, it gave me a lot 👍

nachovizzo, 2024-04-02:
@changh95 @kulfer80 thanks for opening such nice discussions! I will add some bits here.  And, @changh95 about this sentence and your reasoining
>  it's critical to use mid_pose_timestamp for effective KISS-ICP operation.

100 % false in my opinion. I understand your reasoning but here there are 2 options, a) I'm confused or b) you might be overthinking a bit the problem.

Here is a summary of my thoughts

## Why `0.5`

As a matter of fact,  this `0.5` is ONLY THERE to obtain what our community understands as "better results" on the KITTI dataset. 

Why? The reasoning is simple, the so-called "GT" pose from KITTI (which is merely an expensive GPS+IMU system, so it is not ground truth for us) is reporting the GT pose in the middle of the scan. This is not documented anywhere, but it's like so, believe me ;). What probably happened in the history of SLAM is a lot of people were trying motion compensation approaches, without realizing this little fact, and were obtaining probably good results, but the metrics were saying the opposite. Indeed the estimation might be perfect (and go and do yourselves the experiment) but the metrics would be wrong just because one is comparing poses at different time stamps! and thus the error is a conceptual error of the programmer and not an error of the system!

the error of `Odometry_pose_.at(100ms)` ? `GT_pose.at(150ms)`  is not the same as `Odometry_pose_.at(150ms)` ? `GT_pose.at(150ms)`

Sadly, the scientific community is more excited to see bold numbers in tables than to incentivize real research ideas and topic understanding. So, once people saw the bad numbers in the metrics, probably abandoned ideas early on. 

That's why we left that little easter egg in the code, hoping that someday someone curious will raise the flag. So, congratulations!!!! 

## What's really going on?

If you think about it, a typical velodyne-like rotating lidar is not producing **ONE** scan, but rather multiple ones. In general, what's happening inside the hardware is that each beam is being triggered at different times to avoid cross-talk, noise, and so on. So taking the VLP-64 as an example, while the robot is moving the LiDAR extracted `131072` different points (approx) at `131072` different times. This basically means that if we want to perfectly model that movement, we would need to produce `131072` poses per scan, one for each point.

A less crazy approach is to model the "vertical" lines as instantaneous and pretend that the 64 beams in the array were shot at the exact same time. This would give us a total of `2048` poses in our example.

When we want to "deskew" the point cloud, accounting for the fact that there are `2048` different poses while the scan was being captured, what we are doing is trying to approximate the continuous-time scan as an instantaneous measurement of the world. Like if the car would be completely standing still. If we want to do so, we have no other option but to pick just **ONE** of those 2048 moments in time and say : "ok the scan was taken at this moment". When deskewing we are basically transforming the points to that picked location of the scan. Taking the midpoint, just moves all the points to that timestamp without affecting the results. If you take `0.0` or `1.0` the point cloud WILL LOOK THE SAME, and you can have fun and do the experiment. What is different is where this scan is located! which indeed will end up producing a different odometry estimation! just due to the timestamps, but the relative motion would not change at all!

## Conclusion

It does not matter which point you pick to model your Instanenotus scan if it's the first, the middle, or the last point. What matters is what's important for your application. For example, if you are using a robosense lidar in ROS you can instruct the driver to use the very first point it receives as the `stamp` for the message https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/d9dc998a92991a99c5609778691cf3cd47226113/doc/intro/03_hiding_parameters_intro.md?plain=1#L61

This means that when you receive a new PointCloud2 message, the stamp in the header of the message says that it "measured the 360 scan at that exact timestamp". Although we know that the very last point was measured at that timestamp + duration of the total scan. By using this information, then for deskewing, we should be using the `0.0` as the time reference, since we must refer to all the points at the beginning of the scan. Now, with your new deskew pointcloud2 scan, the data makes more sense. Since the stamp in the header says where you "measured" those points, that after correction are modeling a non-moving robot 

I always promised myself to add documentation about this with renders of the clouds to explain, but sadly never found time to do so. I hope I made my point clear with text at least

closed 2024-04-02 by nachovizzo

kulfer80, 2024-04-02:
@nachovizzo Thanks for the detailed explanation and your time.