PRBonn/kiss-icp #304: Question: Has anyone tried KISS-ICP on ARM architecture?

changh95, 2024-03-27:
I noticed that KISS-ICP uses Intel TBB for parallel processing backend.

Considering how Intel CPUs are mostly x86, I am blindly thinking that TBB might not compile on ARM architectures, so not support efficient parallel processing on ARM architectures. Unfortunately, I don't have an ARM system that I can try running KISS-ICP at the moment to test it out. Also, I think there was a major update on TBB a few years ago, which made it difficult to compile new versions of TBB on old versions of OS and compilers, and also to compile old versions of TBB on new versions of OS and compilers. (Ref: https://github.com/compiler-explorer/compiler-explorer/issues/6245 ). Disclaimer though - I don't know deep enough about TBB to say this is 100% true.

This gave me the idea to implement KISS-ICP with OpenMP, which is more compatible than Intel TBB in most system architectures and compilers. (Again, I don't know deep enough about OpenMP to say this is 100% true). Recently I've been learning to use OpenMP and saw the opportunity to implement it. 

Below is my implementation of KISS-ICP with OpenMP. If anyone has an ARM system (e.g. NVIDIA Jetson), I would greatly appreciate if the performance can be tested on it. Also, I would love to get some feedbacks from @nachovizzo - I've always been wondering why TBB was chosen as the parallel processing backend. 

https://github.com/changh95/kiss-icp-omp

nachovizzo, 2024-03-28:
Nice work man !

TBB does compile on ARM architecture. In fact, KISS-ICP runs on Apple Silicon chips! Which are ARM64. So don't worry about that ;) [Here is a nice example of KISS running in ARM64 with TBB enabled.](https://www.linkedin.com/posts/vizzoignacio_roboticssoftware-ros1-ros2-activity-7053335020786049024-P33W?utm_source=share&utm_medium=member_desktop)

> OpenMP, which is more compatible than Intel TBB in most system architectures and compilers

Well.. this is something we can debate. OpenMP generates multithreading code at compile time based on the compiler used, hardware, etc. And all this by using horrible pragmas I don't like at all. So compatibility is not guaranteed at all. If you run the same version compiled on a target platform somewhere else it might now work.

> I've always been wondering why TBB was chosen as the parallel processing backend.

TBB instead is a much richer library making use of real C++ abstractions than OpenMP. OpenMP is "quite old" and way too C style of programming. At KISS we wanted to push a bit the standard programming techniques, based on years of experience with horrible C-style code written in .cpp files from our community. So the decision was made solely in favor of a better style. We never even tried with OpenMP basically because we just don't want to use it ! Additionally, modern C++ standards that accommodate for parallelism are basically wrappers around TBB, so if you want to use `std::execution::par` you will be using TBB under the hood (I wonder why they've picked this instead of OpenMP lol )

And by the way, after you realized that if you want to exploit OpenMP they force you to use C arrays, do you  still believe this is easier to read: 
```cpp

double JTJ_array[36] = {0.0};
double JTr_array[6] = {0.0};

#pragma omp parallel for reduction(+:JTJ_array[:36], JTr_array[:6]) schedule(dynamic) num_threads(NUM_THREADS)
for(size_t i = 0; i < associations.size(); i++) {
    const auto &[J_r, residual] = compute_jacobian_and_residual(associations[i]);
    const double w = GM_weight(residual.squaredNorm());
    Eigen::Matrix6d temp_JTJ = J_r.transpose() * w * J_r;        // JTJ
    Eigen::Vector6d temp_JTr = J_r.transpose() * w * residual;  // JTr
    for (int j = 0; j < 36; j++) {
        JTJ_array[j] += temp_JTJ(j / 6, j % 6);
    }
    for (int j = 0; j < 6; j++) {
        JTr_array[j] += temp_JTr(j);
    }
}

Eigen::Matrix6d JTJ;
Eigen::Vector6d JTr;
for (int j = 0; j < 36; j++) {
    JTJ(j / 6, j % 6) = JTJ_array[j];
}
for (int j = 0; j < 6; j++) {
    JTr(j) = JTr_array[j];
}

```

changh95, 2024-03-29:
Thanks, @nachovizzo, for your insights and sorry for the flood of notifications on GitHub and LinkedIn 😅. I'm heavily using KISS-ICP in my projects and I'm keen to contribute to its community.

Your explanations about OpenMP and TBB were enlightening. Discovering that C++17's pstl uses TBB was an eye-opener for me.

My initial thoughts on OpenMP impl were about how some developers prefer avoiding lambda functions to keep code more portable. For instance, some embedded platforms only support C++98, or need to adhere to strict safety standards like MISRA C++ 2008, or require code that's adaptable to various hardware like CUDA or DSP which forces the base algorithm code to be just simple for-loops. I wondered if using OpenMP's simple pragma for-loop approach might be more suitable in these cases. Below is my code when I first applied OpenMP, and to me it looked much easier to read, and also to change in the future.

```cpp
  LinearSystem J(Eigen::Matrix6d::Zero(), Eigen::Vector6d::Zero());

  #pragma omp parallel for num_threads(NUM_THREADS)
  for(size_t i = 0; i < associations.size(); i++) {
      const auto &[J_r, residual] = compute_jacobian_and_residual(associations[i]);
      const double w = GM_weight(residual.squaredNorm());
      LinearSystem temp(J_r.transpose() * w * J_r,        // JTJ
                        J_r.transpose() * w * residual);  // JTr
      #pragma omp critical
      {
          J.first += temp.first;
          J.second += temp.second;
      }
  }

  return J;
```

However, your points made me see the benefits of TBB, especially in contexts where its features outshine OpenMP's simplicity. I'm grateful for your perspective—it's given me a lot to think about and has definitely broadened my understanding.

Moreover, I'm considering experimenting with replacing TBB with C++17 pstl or even CUDA to see how it impacts performance and portability. Once I achieve some promising results, I plan to share my findings with the community :smile:

changh95, 2024-03-29:
Since the question is answered, I'll close the issue.
Thanks again!

closed 2024-03-29 by changh95