PRBonn/kiss-icp #315: Cannot recreate the KITTI results

eugeniu1994, 2024-04-05:
Hello,

Thank you for sharing your work.

I used KISS-ICP for LIDAR odometry on the KITTI dataset.
The motion correction was disabled and the provided kitti_correction was used.
I compared against the GPS ground truth and the semantic Kitti poses,  using the absolute and relative Metrics provided in this project,  however, I got the average relative translation error of 0.6% different from the 0.49% mentioned in the paper.

Is there something else you did to get that result?

Regards!

nachovizzo, 2024-04-05:
Semantic kitti poses are not GT poses

closed 2024-04-05 by nachovizzo

eugeniu1994, 2024-04-05:
Against the GPS poses the relative accuracy was 0.6% for the first 10 KITTI sequences.
Is there something else that should be done for KITTI @nachovizzo  to get the same results? 

Regards!

nachovizzo, 2024-04-05:
You can try with the version that was used for the paper almost 2 years ago https://github.com/PRBonn/kiss-icp/releases/tag/v0.0.13. But I'd say if you are getting those results something strange might be happening on your side. You can also check the files with the trajectories used when we evaluated on Kitti here https://github.com/PRBonn/kiss-icp/tree/main/eval

In any case it's likely that you are reading another file or tagt we have a Regression in our code