PRBonn/kiss-icp #317: Deskew in Config is erased by not passing --deskew in kiss_icp_pipeline

ClementLeBihan, 2024-04-08:
Hi,
When calling kiss_icp_pipeline with a config file that contains deskew: True (`kiss_icp_pipeline input.bag --topic /scan --config config.yaml`), the deskew parameter is erased to False as OdometryPipeline is called with the deskew parameter from argument parser. So if I do not call kiss_icp_pipeline with the --deskew flag, it don't use the deskew parameter in config file. 

Is it normal ? 
Is there any other config option that are erased ?

nachovizzo, 2024-04-08:
@ClementLeBihan are you sure about this ? could you please provide more details ? In my case is working.

PS: please post the config file

ClementLeBihan, 2024-04-08:
If it works on your case I'm not sure anymore, but when I print the deskew parameter in OdometryPipeline init function it is false, so `load_config(config, deskew=deskew, `max_range=max_range)` is called with `deskew=False` and deskew is overidden to False in load_config function ... 
When I add `print("before overriding config : ", config)` and `print("after overriding config : ", config)` in load_config.py here is what I go : 
```Before overriding config :  out_dir='results' data=DataConfig(preprocess=True, max_range=100.0, min_range=4.0, deskew=True) mapping=MappingConfig(voxel_size=0.5, max_points_per_voxel=50) adaptive_threshold=AdaptiveThresholdConfig(fixed_threshold=None, initial_threshold=2.0, min_motion_th=0.1)
After overriding config :  out_dir='results' data=DataConfig(preprocess=True, max_range=100.0, min_range=4.0, deskew=False) mapping=MappingConfig(voxel_size=0.5, max_points_per_voxel=50) adaptive_threshold=AdaptiveThresholdConfig(fixed_threshold=None, initial_threshold=2.0, min_motion_th=0.1)```
And my pointcloud isn't deskewed ...

Here is my config file : 
```out_dir: "results"

data:
  deskew: True
  preprocess: True
  max_range: 100.0
  min_range: 4

mapping:
  voxel_size: 0.5
  max_points_per_voxel: 50```

nachovizzo, 2024-04-08:
Have you checked the data? That would be the most important thing. If you see the cloud being deskew or not? If you see perfect circles even when there is motion they it's not being deskewed. I'll check into this anyway

ClementLeBihan, 2024-04-08:
Yes I've checked the data and the pointcloud isn't deskew unless I add the flag --deskew. 
Without --deskew flag, get_motion_compensator return StubCompensator()

benemer, 2024-04-11:
This PR https://github.com/PRBonn/kiss-icp/pull/305 should have solved this. Can you build KISS from source and check again?

benemer, 2024-04-11:
As soon as @nachovizzo pulls the trigger for a new release :)

We are currently changing quite a few things in another PR, so a new release is very likely to arrive soon!

nachovizzo, 2024-04-11:
Thanks @benemer .
@ClementLeBihan it will take some time until the next release, in the meantime you can also run this very beautiful command to install that particular fix, or change it for `main` in case you want the latest
```
pip install git+https://github.com/PRBonn/kiss-icp.git@11c7d97d156a7ff72fefdacc3105db4a99805e0e#subdirectory=python
```