PRBonn/kiss-icp #334: Build is broken

mattwilliamson, 2024-04-30:
Building on humble. 0.4.0 works, but from head as of today, I get the following error: `error: ‘tbb::info’ has not been declared`

```
11490.2 Starting >>> kiss_icp
11509.4 --- stderr: kiss_icp
11509.4 CMake Deprecation Warning at /opt/ros/humble/build/kiss_icp/_deps/sophus-src/CMakeLists.txt:1 (cmake_minimum_required):
11509.4   Compatibility with CMake < 3.5 will be removed from a future version of
11509.4   CMake.
11509.4 
11509.4   Update the VERSION argument <min> value or use a ...<max> suffix to tell
11509.4   CMake that the project does not need compatibility with older versions.
11509.4 
11509.4 
11509.4 CMake Deprecation Warning at /opt/ros/humble/build/kiss_icp/_deps/tessil-src/CMakeLists.txt:1 (cmake_minimum_required):
11509.4   Compatibility with CMake < 3.5 will be removed from a future version of
11509.4   CMake.
11509.4 
11509.4   Update the VERSION argument <min> value or use a ...<max> suffix to tell
11509.4   CMake that the project does not need compatibility with older versions.
11509.4 
11509.4 
11509.4 CMake Warning:
11509.4   Manually-specified variables were not used by the project:
11509.4 
11509.4     HUNTER_KEEP_PACKAGE_SOURCES
11509.4 
11509.4 
11509.4 /opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp: In constructor ‘kiss_icp::Registration::Registration(int, double, int)’:
11509.4 /opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp:174:69: error: ‘tbb::info’ has not been declared
11509.4   174 |       max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::info::default_concurrency()) {
11509.4       |                                                                     ^~~~
11509.4 make[2]: *** [kiss_icp/core/CMakeFiles/kiss_icp_core.dir/build.make:76: kiss_icp/core/CMakeFiles/kiss_icp_core.dir/Registration.cpp.o] Error 1
11509.4 make[1]: *** [CMakeFiles/Makefile2:322: kiss_icp/core/CMakeFiles/kiss_icp_core.dir/all] Error 2
11509.4 make: *** [Makefile:146: all] Error 2
11509.4 ---

```

nachovizzo, 2024-04-30:
:thinking: could you provide more details about your situation? Like why do you have C++ file in `opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp` ???

You likely have mixup between new/old versions of the system as `tbb::info` is literally explicitly included in that translation unit:
https://github.com/PRBonn/kiss-icp/blob/cc8d18ecc50f2c4b56ede18fa16c9a9b0a5a9451/cpp/kiss_icp/core/Registration.cpp#L27

So the only chance here is that 
> from head as of today,

is not really your case

mattwilliamson, 2024-05-01:
Good question! Thank you for pointing that out. I am looking further, but it seems like a conflict with `rtabmap_ros`. 

```sh
# find / -name Registration.cpp
/opt/ros/humble/src/rtabmap/corelib/src/Registration.cpp
/opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp

# diff /opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp /opt/ros/humble/src/kiss-icp/cpp/kiss_icp/core/Registration.cpp
#
```

I'll try removing that package and see if it helps.

mattwilliamson, 2024-05-01:
Just to add more context, I'm building ROS2 humble in a docker container because I'm on Jetson Orin Nano and need to build from source. I'm doing something like this:

`deepdrive.repos`
```yaml
  kiss-icp:
    type: git
    url: https://github.com/PRBonn/kiss-icp.git
    version: main
    # version: v0.4.0
```

```sh
vcs import src < deepdrive.repos

DEPS="rtabmap_ros joint_state_publisher nav2_bringup navigation2 rmw_cyclonedds_cpp rmw_fastrtps_cpp robot_localization robot_state_publisher ros2bag rosbag2_storage_default_plugins rqt_tf_tree slam_toolbox twist_mux usb_cam xacro foxglove_msgs gscam "

rosinstall_generator --deps --rosdistro ${ROS_DISTRO} $DEPS > ros2.${ROS_DISTRO}.rosinstall
vcs import src < ros2.${ROS_DISTRO}.rosinstall
apt-get update && rosdep update
rosdep install -i --from-path ./ --ignore-src -r -y --rosdistro $ROS_DISTRO --skip-keys "$SKIP_KEYS"
colcon build --merge-install \
    --cmake-args -DCMAKE_BUILD_TYPE=Release \
    --cmake-args -DBUILD_TESTING=OFF \
    --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
    --cmake-args -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
    --cmake-args -DBUILD_SHARED_LIBS=ON \
    --cmake-args -DHUNTER_KEEP_PACKAGE_SOURCES=ON
```

nachovizzo, 2024-05-01:
It shouldn't be a problem that two files with the same name under different packages Co exists. Do you have installed libtbb-dev? Could you please post the version.? The kiss icp ROS wrapper is missing some dependencies on the package.xml file I'll add soon

nachovizzo, 2024-05-01:
@mattwilliamson could you try nacho/add_missing_deps_to_package_xml branch to see if this changes anything?, basically change your .repos for 
```
  kiss-icp:
    type: git
    url: https://github.com/PRBonn/kiss-icp.git
    version: nacho/add_missing_deps_to_package_xml branch
```

mattwilliamson, 2024-05-02:
Certainly!

mattwilliamson, 2024-05-02:
Same error unfortunately.

```  
kiss-icp:
    type: git
    url: https://github.com/PRBonn/kiss-icp.git
    # version: main
    version: nacho/add_missing_deps_to_package_xml
```

nachovizzo, 2024-05-02:
> Same error unfortunately.
> 
> ```
> kiss-icp:
>     type: git
>     url: https://github.com/PRBonn/kiss-icp.git
>     # version: main
>     version: nacho/add_missing_deps_to_package_xml
> ```
> 
> It seems like a versioning issue with tbb, right?

I don't think so honestly... that header has been there for more than 4 years, and the error you get is at compilation time. Something more fishy is going on, but without access to the target system, I can't see.  You can rest assured that the kiss icp pipeline has been successfully built in many platforms and applications, and also in the CI/CD of this repository. So the build is certainly guaranteed to succeed

nachovizzo, 2024-05-02:
You can build with `VERBOSE=1` and also try to fetch the `compile_commands.json` to see the effective build line. Maybe the tbb headers are not being included properly, I don't know

mattwilliamson, 2024-05-03:
Thank you for your help. I'll dig some more.

barcesat, 2024-05-09:
Hi,
I also have the same problem. I'm trying toi build kiss-icp from source on ros humble Jetson Orin Nano.
I had a duplicate Registration.cpp file from RTABMAP source but I've deleted it.

Short version:
```shell
$ colcon build --event-handlers console_cohesion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON --packages-select kiss_icp

/home/peterpan/mini_ws/src/mapping/kiss-icp/cpp/kiss_icp/core/Registration.cpp: In constructor ‘kiss_icp::Registration::Registration(int, double, int)’:
/home/peterpan/mini_ws/src/mapping/kiss-icp/cpp/kiss_icp/core/Registration.cpp:174:69: error: ‘tbb::info’ has not been declared
  174 |       max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::info::default_concurrency()) {
```

Long version:

```shell
$ colcon build --event-handlers console_cohesion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON --packages-select kiss_icp
Starting >>> kiss_icp
[Processing: kiss_icp]                             
--- output: kiss_icp                               
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Turning basic logging ON
-- Found ament_cmake: 1.3.6 (/opt/ros/humble/share/ament_cmake/cmake)
-- Found Python3: /usr/bin/python3.9 (found version "3.9.5") found components: Interpreter 
-- Found geometry_msgs: 4.2.3 (/opt/ros/humble/share/geometry_msgs/cmake)
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found nav_msgs: 4.2.3 (/opt/ros/humble/share/nav_msgs/cmake)
-- Found rclcpp: 16.0.7 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.5 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1f")  
-- Found FastRTPS: /opt/ros/humble/include  
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found rclcpp_components: 16.0.7 (/opt/ros/humble/share/rclcpp_components/cmake)
-- Found sensor_msgs: 4.2.3 (/opt/ros/humble/share/sensor_msgs/cmake)
-- Found tf2_ros: 0.25.5 (/opt/ros/humble/share/tf2_ros/cmake)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp
/usr/bin/cmake -S/home/peterpan/mini_ws/src/mapping/kiss-icp/ros -B/home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp/CMakeFiles /home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp'
/usr/bin/make -f CMakeFiles/kiss_icp_node.dir/build.make CMakeFiles/kiss_icp_node.dir/depend
[... 185 lines cut ...]
cd /home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp/kiss_icp/metrics && /usr/bin/cmake -P CMakeFiles/kiss_icp_metrics.dir/cmake_clean_target.cmake
cd /home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp/kiss_icp/metrics && /usr/bin/cmake -E cmake_link_script CMakeFiles/kiss_icp_metrics.dir/link.txt --verbose=1
/usr/bin/ar qc libkiss_icp_metrics.a  CMakeFiles/kiss_icp_metrics.dir/Metrics.cpp.o
/usr/bin/ranlib libkiss_icp_metrics.a
make[2]: Leaving directory '/home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp'
[ 64%] Built target kiss_icp_metrics
make[1]: Leaving directory '/home/peterpan/mini_ws/src/mapping/kiss-icp/build/kiss_icp'
make: *** [Makefile:144: all] Error 2
---
--- stderr: kiss_icp
/home/peterpan/mini_ws/src/mapping/kiss-icp/cpp/kiss_icp/core/Registration.cpp: In constructor ‘kiss_icp::Registration::Registration(int, double, int)’:
/home/peterpan/mini_ws/src/mapping/kiss-icp/cpp/kiss_icp/core/Registration.cpp:174:69: error: ‘tbb::info’ has not been declared
  174 |       max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::info::default_concurrency()) {
      |                                                                     ^~~~
make[2]: *** [kiss_icp/core/CMakeFiles/kiss_icp_core.dir/build.make:66: kiss_icp/core/CMakeFiles/kiss_icp_core.dir/Registration.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:341: kiss_icp/core/CMakeFiles/kiss_icp_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:144: all] Error 2
---
Failed   <<< kiss_icp [34.8s, exited with code 2]

Summary: 0 packages finished [35.4s]
  1 package failed: kiss_icp
  1 package had stderr output: kiss_icp
```

Grimwork, 2024-05-23:
Hi,
Same problem here when i try to build on a Jetson Orin.
Has anyone solved this ?

nachovizzo, 2024-05-23:
Re opening for investigation

SebsBarbas, 2024-06-02:
Seems like a problem with the latest commits. I have been trying to install it on my laptop (Intel based, Ubuntu 20.04) and the it did not build, with the same error message as mentioned by the others commenting, using the `make editable` command. I have rolled back to version 0.3.0 and it has build first try. 

Please let me know if I can help somehow with more evidence or whatever is needed! If needed with git bisect I can also help!

kulfer80, 2024-06-05:
You can mby try this temp. fix (comment tbb/info.h and replace it with tbb/tbb.h):
```cpp
// #include <tbb/info.h>
#include <tbb/tbb.h>
max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::task_scheduler_init::default_num_threads()) 
 ```

mehermvr, 2024-06-26:
> -- The CXX compiler identification is GNU 9.4.0

Noticing this, because earlier I had run into similar problems with TBB when trying to compile on an Ubuntu 20.04 docker image, I believe the problem can be the libtbb version.
I've put together a minimal dockerfile possibly replicating the issue:

```docker
FROM ubuntu:20.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get upgrade -y \
  && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt-get install --no-install-recommends -y \
  git \
  build-essential \
  cmake \
  ca-certificates \
  && rm -rf /var/lib/apt/lists/*

WORKDIR /workspace

RUN git clone https://github.com/PRBonn/kiss-icp.git

WORKDIR /workspace/kiss-icp

# with the following commented, build is fine
# RUN apt-get update && apt-get install --no-install-recommends -y libtbb-dev && rm -rf /var/lib/apt/lists/*

RUN make cpp
```

Uncommenting the install libtbb-dev line and running again throws the following error (i've shortened the full build log)
```
0.461 -- The CXX compiler identification is GNU 9.4.0                                                                       
0.467 -- Check for working CXX compiler: /usr/bin/c++                                                                       
0.557 -- Check for working CXX compiler: /usr/bin/c++ -- works                                                              
0.558 -- Detecting CXX compiler ABI info                                                                                    
0.650 -- Detecting CXX compiler ABI info - done
0.666 -- Detecting CXX compile features
0.666 -- Detecting CXX compile features - done
1.811 -- The C compiler identification is GNU 9.4.0
1.813 -- Check for working C compiler: /usr/bin/cc
1.889 -- Check for working C compiler: /usr/bin/cc -- works
1.890 -- Detecting C compiler ABI info
1.967 -- Detecting C compiler ABI info - done
1.977 -- Detecting C compile features
1.977 -- Detecting C compile features - done
.....bunch of stuff later.....
8.453 Building CXX object core/CMakeFiles/kiss_icp_core.dir/Deskew.cpp.o
8.453 Building CXX object core/CMakeFiles/kiss_icp_core.dir/VoxelHashMap.cpp.o
9.575 /workspace/kiss-icp/cpp/kiss_icp/core/Registration.cpp: In constructor 'kiss_icp::Registration::Registration(int, double, int)':
9.575 /workspace/kiss-icp/cpp/kiss_icp/core/Registration.cpp:174:69: error: 'tbb::info' has not been declared
9.575   174 |       max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::info::default_concurrency()) {
9.575       |                                                                     ^~~~
11.97 make[3]: Leaving directory '/workspace/kiss-icp/build'
11.97 make[3]: *** [core/CMakeFiles/kiss_icp_core.dir/build.make:63: core/CMakeFiles/kiss_icp_core.dir/Registration.cpp.o] Error 1
11.97 make[2]: *** [CMakeFiles/Makefile2:33739: core/CMakeFiles/kiss_icp_core.dir/all] Error 2
11.97 make[2]: *** Waiting for unfinished jobs....
19.61 Linking CXX static library libkiss_icp_metrics.a
19.64 make[3]: Leaving directory '/workspace/kiss-icp/build'
19.65 Built target kiss_icp_metrics
19.65 make[2]: Leaving directory '/workspace/kiss-icp/build'
19.65 make[1]: *** [Makefile:130: all] Error 2
19.65 make[1]: Leaving directory '/workspace/kiss-icp/build'
19.65 make: *** [Makefile:18: cpp] Error 2
```

The version of tbb installed is `Version: 2020.1-2` (latest available as per ubuntu package repos).
Meanwhile, kiss fetches v2021.8.0 when using FetchContent, so this could be the problem. 

> You can mby try this temp. fix (comment tbb/info.h and replace it with tbb/tbb.h):
> 
> ```c++
> // #include <tbb/info.h>
> #include <tbb/tbb.h>
> max_num_threads_(max_num_threads > 0 ? max_num_threads : tbb::task_scheduler_init::default_num_threads()) 
> ```

I tried this one, but it doesn't help at least here. 

I'm not familiar with the Jetson Orion and its build env but maybe a potential solution is to just uninstall system tbb and try again?

tizianoGuadagnino, 2024-07-09:
@mehermvr thanks for the help here, I close this due to inactivity.

nachovizzo, 2024-07-09:
@tizianoGuadagnino I was leaving this open to analyze the possibly to include a jetson-like docker in our build to make sure we also support this platform today and in the future

nachovizzo, 2024-07-09:
Could anyone test if channging this `tbb::info::default_concurrency()` line https://github.com/PRBonn/kiss-icp/blob/e19823ab7275199caee037d75c6a3c7d6945a628/cpp/kiss_icp/core/Registration.cpp#L174 to 
```
this_task_arena::max_concurrency()
```

fix the build ?

nachovizzo, 2024-07-09:
@mattwilliamson , @SebsBarbas , @barcesat , @Grimwork 

I have a fix ready in https://github.com/PRBonn/kiss-icp/pull/361, sorry for the delay . I only managed to fix this thanks to the hint of @SebsBarbas as before him no one mentioned that Jetson are running ubuntu 20.04 flavored distros, and that's the real problem ! I now added a dev CI for 20.04 to make sure this won't happen in the future again

PR #337 (2024-05-03): Cleanup ROS package.xml and CMakeLists.txt
Add missing system dependencies as well

Potentially fixes #334
closed 2024-07-10 by nachovizzo