PRBonn/kiss-icp #346: directory file not found

Sefilevy, 2024-06-06:
i tried with two different kitti_raw datasets:
- 2011_09_30/2011_09_30_drive_0016_sync
- 2011_10_03/2011_10_03_drive_0042_sync
in both cases i got the same error about the connectivity between the IMU and the velodyne:
"FileNotFoundError: [Errno 2] No such file or directory: 
'2011_09_30/2011_09_30_drive_0016_sync/2011_09_30/calib_imu_to_velo.txt'"
i checked in the folders and i didn't find this file although you mention those sequences as valid for the kiss-icp

thank you for your help!

Sefilevy, 2024-06-07:
i downloaded the calibration zip also- do i need to add the specific file of the calib_IMU_velo to the folder of the dataset?

benemer, 2024-06-14:
Yes, you have to download and add it by yourself

closed 2024-06-14 by benemer