PRBonn/kiss-icp #357: Any possibility / benefit to add temporal limits to the local map

saadehmd, 2024-07-03:
First of all, a bundle of thanks and appreciation for the authors , to provide such an extensive implementation available openly and very nice documentation and support on issues as well. So i have been recently looking into the utility of KISS-ICP for radar data. I am familiar with some radar-specific methods like https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/casado-herraez2024icra.pdf (Eagerly waiting for it's opensource availability in near future.) For now i resorted to using a combination of KISS-ICP and external fused-odometry( for initial_pose). Following videos show some nice results (albeit some jerky motion.) I'm also looking into possibly using the EKF-LIO extension to KISS_ICP (because this odometry source might not always be too reliable for my use-case.)

Anyways, my question is, is it possible to introduce a "temporal limit" to the accumulated local_map: i.e.; The local map only has voxels populated from of last N frames? I know that this hasn't been explicitly studied in the paper or provided as a config in the code, but i'm both curious to know whether this can be done and should this be done ? Would it benefit to have local_map restricted only to N number of previous frames? 

I recorded some videos that demonstrate the accumulation of "undesirable artifacts" stored as a part of map due to noisy radar data (flares, multipath-reflections, smeared clouds of solid objects). These kind of artifacts are probably inherent to the radar pointclouds and that's probably why  a lot radar methods have additional ways to filter data (whether temporally or by pruning the input cloud.). 

Moreover, even if we talk about LIDARs only, the moving objects in the scene will always create these unwanted ghost-voxels, which can easily fill-up the local map(at worse) or create clutter around the actually interesting landmarks(at best).  It's the same problem as the noise except the "noise" in this context are the moving objects. I get that the local_map KISS_ICP uses, is not actually a probabalistically updated spatio-temporal map (like an Octomap) but is rather just accumulating points over time.

But wouldn't it actually help to somehow also prune/erase from the voxel grid when certain points in the grid have not recieved any hits for N number of frames? 

https://github.com/PRBonn/kiss-icp/assets/62055851/929f6866-07f2-4588-be3c-8a05f8c8e8da

https://github.com/PRBonn/kiss-icp/assets/62055851/929f6866-07f2-4588-be3c-8a05f8c8e8da

https://github.com/PRBonn/kiss-icp/assets/62055851/929f6866-07f2-4588-be3c-8a05f8c8e8da

https://github.com/PRBonn/kiss-icp/assets/62055851/9d13f725-8e5f-425b-bf15-309b832ac40b

https://github.com/PRBonn/kiss-icp/assets/62055851/9d13f725-8e5f-425b-bf15-309b832ac40b

https://github.com/PRBonn/kiss-icp/assets/62055851/9d13f725-8e5f-425b-bf15-309b832ac40b

https://github.com/PRBonn/kiss-icp/assets/62055851/81dbf7ec-cce9-458a-a2a0-1dda7586f93a

https://github.com/PRBonn/kiss-icp/assets/62055851/81dbf7ec-cce9-458a-a2a0-1dda7586f93a

https://github.com/PRBonn/kiss-icp/assets/62055851/81dbf7ec-cce9-458a-a2a0-1dda7586f93a

benemer, 2024-07-18:
We modified the voxel hash map for our MapMOS project to keep the timestamp when the points were added to the map; see [here](https://github.com/PRBonn/MapMOS/blob/main/src/mapmos/pybind/VoxelHashMap.hpp).

With this implementation, you can use the `PointcloudWithTimestamps()` method to get the points and filter them based on the timestamp. Or, you can modify the `RemoveFarAwayPoints()` method to remove points that are too old.

We also maintain a belief state in this modified voxel hash map that can be used to model a voxel's dynamic state.

benemer, 2024-07-23:
I will close this for now, feel free to re-open if anything new pops up :)

closed 2024-07-23 by benemer