PRBonn/kiss-icp #378: deb release?

MCFurry, 2024-07-26:
Thanks a lot for this great package! We've been playing around with it some time and really like it!

Would it be possible to release kiss-icp to the ROS buildfarm and make this package apt installable?

nachovizzo, 2024-07-26:
@MCFurry, this is definetly on my todo list. But have you released a package in the ROS buildfarms in the past? It is one of the most annoying and painful tasks I have done in recent years.

Out of curiosity, are you using this on a mobile platform???

MCFurry, 2024-07-26:
It shouldn't be too much of an issue, my colleague can assist you in doing so if you like?

However we should check if the current package layout is sufficient or if we better change that before blooming.

We're not using it on a mobile platform runtime yet, but we've been experimenting with mobile robot datasets offline quite a bit.

nachovizzo, 2024-07-29:
@MCFurry, I was asking because (spoiler alert) we are currently working on the next generation of kiss-icp for wheeled mobile robots. The new system makes kiss-icp look ridiculous, and I'm still wondering if it's worth releasing kiss-icp as a ROS package (and going through the bloom-pain: I just released a python package, and it took me 2 hours) or waiting for the new generation of the system and release that one.

I'm open to this, so if you believe that is still worth releasing kiss icp as a ROS package I can go ahead and do it

MCFurry, 2024-07-29:
I see, that sounds promising! Do you have an expected timeline on finalizing your new package?

nachovizzo, 2024-07-29:
It's under heavy development, but we still need to tweak the last bits, so we don't have an expected release time. But I would assume it would be ready by the end of this year, I will notify you in this issue so you are aware

MCFurry, 2024-07-29:
Cool, thanks for that and good luck finishing your new package!

closed 2024-07-30 by tizianoGuadagnino