PRBonn/kiss-icp #384: kiss_icp_pipeline and lidar_visualizer error

kanhw, 2024-08-06:
when I use: kiss_icp_pipeline --dataloader kitti --sequence 00 --visualize dataset/
it shows:
[image]
The same to lidar_visualizer.Do you have any good solutions?

nachovizzo, 2024-08-06:
Is not an error just a warning, we did fix it in lidar visualizer but won't fix for kiss icp as we have a new visualizer ready to deploy very soon https://github.com/PRBonn/kiss-icp/pull/379

In any case, this is because you have a newer version of open3d, if it relly bothers you you can downgrade `pip install open3d==0.17`

closed 2024-08-06 by nachovizzo

kanhw, 2024-08-06:
Thank you for your reply, sorry to bother you again, I changed the version of open3d to 0.17, this warning will no longer appear, but the visualization will only show a single frame of the radar point cloud, the terminal displayes as shown in the figure.
[image]