PRBonn/kiss-icp #386: Adaptive thresholding

gassercariad, 2024-08-06:
I just used your repo in a project, so I am not expert in working with lidar. But, I have a question regarding how this adaptive threshold works, so my first question after debugging your code, apparently what you initalize as 2 is the sigma, and not the tau (as mentioned in the paper) can you confirm this?

I tested kiss-icp, in a fast moving environment (car on the highway), the threshold was decreasing quickly and getting a completely wrong output, while increasing the initial threshold or even using a fixed one was much better? I believe the adaptive thresholding can be done in a better way to also take into consideration the previous speed itself?

tizianoGuadagnino, 2024-08-09:
Hi, as it is written in the paper $\tau = 3 \sigma$, which is what it is implemented in the code [here](https://github.com/PRBonn/kiss-icp/blob/95f6dfdf54cae8eaecdb7fa66c09e0b0fa0489b6/cpp/kiss_icp/pipeline/KissICP.cpp#L62).

I do not understand what you mean exactly by _completely wrong output_, but I am very open to improving the adaptive threshold, so if you have ideas, feel free to brain-storm on a PR. I am not sure that the previous velocity is going to give any further insights for the threshold, but I will like to see the failure case that you describe.

tizianoGuadagnino, 2024-08-09:
Yes, you are right; this was done to simplify the implementation a bit; in the long run, you should converge to the same value of $\tau$ anyway. Take into account that that paper is now 2 years behind the current implementation ;)

Let me know if you find out anything about the threshold, for now I close the issue and wait for news from your side.

closed 2024-08-09 by tizianoGuadagnino