PRBonn/kiss-icp #399: result npy, result mapping?

loleklel, 2024-09-18:
Hi,
I'm trying your kiss icp algorithm, at first sight it looks good in the visualizer, but I'm a bit confused about the output in the results ordner.

Can you give me more information about the files and structure, what it means and how can I get the point cloud mapping as a result?

All I get is:
poses.npy, poses_kitti.txt, poses_tum.txt

Cheers,
Robin

loleklel, 2024-09-23:
well, I finally run kiss icp with ros2 and can get local map....

benemer, 2024-09-25:
KISS-ICP is not a mapping pipeline. It estimates the LiDAR odometry. That is why in the results folder you will have the estimated poses in different data formats.

If you need the local map, you must modify the code to dump it to disk or use ROS 2 and subscribe to the topic. However, remember that this local map is a downsampled and cropped version of the aligned scans used to align the current scan. It is not a map in the sense of a global map for localization for example.

closed 2024-09-25 by benemer