PRBonn/kiss-icp #403: Testing KISS-ICP in ROS2

mppark40, 2024-10-13:
I have converted KITTI dataset in ros2 bag file so i could use in ros2 launch. Then i used subscribed the kiss/odometry topic so i could use this text file and compare with the ground truth of the scenarios in KITTI. However the RMSE performance was not that good. Did i expeerimented the wrong way?
1. Converted the KITTI dataset to a ROS2 bag file for use in a ROS2 launch.
2. Subscribed to the kiss/odometry topic.
3. Exported the topic data to an Excel file, then converted it to a text file for comparison with the ground truth from the KITTI dataset scenarios.
4. However, the RMSE and other performances were not as good as expected.
Is there something wrong with my experimental approach?

tizianoGuadagnino, 2024-10-14:
Why do you want to convert the KITTI dataset to a rosbag? Just use the Kitti dataloader that we provide with the python API. We do not guarantee performance if you, by any means, manipulate the data by converting it.

horverno, 2024-10-14:
It should run smoothly. Check this: https://autowarefoundation.github.io/autoware-documentation/main/support/troubleshooting/performance-troubleshooting/ it is autoware but mostly applies to ros 2 as well.

closed 2024-10-16 by tizianoGuadagnino

mppark40, 2024-10-18:
> Why do you want to convert the KITTI dataset to a rosbag? Just use the Kitti dataloader that we provide with the python API. We do not guarantee performance if you, by any means, manipulate the data by converting it.

I'm a master's student working on LiDAR odometry and I got into KISS-ICP. I'm trying to create a new algorithm that combines IMU data. I tested a simple version using the ROS2 KITTI dataset and created a loosely-coupled odometry algorithm using the IMU topic. I ran the existing KISS-ICP ROS2 launch on the KITTI dataset to compare, but the performance wasn't great.