PRBonn/kiss-icp #435: TUM RGBD problems

cosminvlaicu, 2025-01-28:
Should the data work well with TUM RGBD?

In order to run the dataset, I root caused an error, which emptied depth frames:

replaced

    def preprocess(self, frame: np.ndarray, timestamps: np.ndarray, relative_motion: np.ndarray):
        return np.asarray(
            self._preprocessor._preprocess(
                kiss_icp_pybind._Vector3dVector(frame),
                timestamps.ravel(),
                relative_motion,
            )
        )

with

    def preprocess(self, frame: np.ndarray, timestamps: np.ndarray, relative_motion: np.ndarray):
        return np.asarray(
            kiss_icp_pybind._Vector3dVector(frame)
        )

Nevertheless, the trajectory is visually awful. Is this expected?

cosminvlaicu, 2025-01-28:
I installed the latest version manually and the bug has been removed, so you can disregard my patch. Nevertheless, the TUM results look very bad on https://cvg.cit.tum.de/rgbd/dataset/freiburg1/rgbd_dataset_freiburg1_floor.tgz

[kissicp_tumrgbd_result.txt](https://github.com/user-attachments/files/18566599/kissicp_tumrgbd_result.txt)

tizianoGuadagnino, 2025-01-28:
If you want to use our system on an RGB-D dataset, you probably need to perform some more tuning than simply running the pipeline. Our system is LiDAR odometry not Point Cloud Odometry. Nonethenless, I can provide you with a config that surely works better then default.

[kiss_icp.txt](https://github.com/user-attachments/files/18571035/kiss_icp.txt)

I will close the issue anyways ;)