PRBonn/kiss-icp #447: Deskew point selection

Samahu, 2025-03-17:
Regarding the PR: https://github.com/PRBonn/kiss-icp/pull/427

When upgraded to version 1.2+ we noticed a degradation in the quality of the PointCloud that we generate through KissICP, tracing things down it appears the degradation occurred around the point where you moved the deskew point from mid to start.

I read through the discussion and I understand the motivation for this change, however, this change does affect the outcome of KissICP.

Do you think it would make sense to make the deskew a parameter of the **Preprocessor** that the caller can assign the value within the range [0.0, 1.0]?

benemer, 2025-03-18:
What kind of degradation do you observe? Can you share some of the results or data so that we can examine why the performance dropped?

Samahu, 2025-03-18:
Here is SLAM output when using KissICP 1.1

[image]

Here is the result after upgrading to KissICP 1.2+

[image]

benemer, 2025-03-21:
Have you tried changing back the deskewing to the midpoint? Just to ensure that no other change between v1.1 and v1.2 introduced this. Also, can you verify that deskewing is on for both cases?

I am unsure if adding this as an optional parameter is the way to go because it does not make sense to deskew with respect to other timestamps when using the constant velocity model. Instead, I would like to see if something else causes this effect.

Samahu, 2025-03-21:
> Also, can you verify that deskewing is on for both cases?

I had clone the kiss-icp repo (v1.2.2) and enabled editable and then changed this back to 0.5, then I was able to achieve the previous results.

benemer, 2025-03-24:
Okay, that's interesting. Could you share the data with us so we can also look at it?

Samahu, 2025-03-24:
Here: https://static.ouster.dev/sdk-docs/0.5.1/sample-data.html

benemer, 2025-03-25:
And which sequence?

Also, how do you get the visualization? Is that the `ouster-cli`? Just post which command you are running so that I can give it a try.

Samahu, 2025-03-25:
I think any sequence would show the difference, let's pick the first one for example.

Using the current ouster-cli (version 0.14.0 which depends on kiss-icp 1.1) if you use the commands:
```bash
ouster-cli source OS0-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap slam save stable.osf
ouster-cli source stable.osf save stable.pcd
```

Next modify kiss-icp version 1.1 `Preprocessor::deskew` method to match the latest changes in 1.2.2 and re-run
```bash
ouster-cli source OS0-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap slam save current.osf
ouster-cli source current.osf save current.pcd
```

Observe the difference:
```bash
$ ls -lh | grep pcd
74M Mar 25 08:30 current-000.pcd
72M Mar 25 07:47 stable-000.pcd
```

The `OS0-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap` is a short sequence of only 201 frames, the difference in pcd size is more pronounced as the sequence size is increased.

> Note: I suggested modifying the kiss-icp 1.1 code since you can test it directly against ouster-sdk 0.14 without making changes to ouster-sdk code but I have produced similar results after updating our mapping code to 1.2.2 and although we do see notable improvements in 1.2.2 vs 1.1.0 the point cloud quality has degraded.

Please let us know if you are able to re-produce the results or need more details.

Thanks!

Samahu, 2025-03-25:
also to visualize a sequence simply append the viz command: `ouster-cli source stable.osf viz`

Samahu, 2025-03-26:
> If you achieved the previous results usingt he ouster-sdk, I think it is not possible that you used kiss-icp==1.2.2, because the ouster-sdk strictly enforces only v1.1.0.

I have tested on a development branch of ouster-sdk with kiss-icp 1.2.2 enabled.

> > modify kiss-icp version 1.1 Preprocessor::deskew
> 
> This version does not have the `Preprocessor` class. I am a bit confused what exactly you tested to achieve the results posted above.

I was referring to this method https://github.com/PRBonn/kiss-icp/blob/v1.1.0/cpp/kiss_icp/core/Preprocessing.hpp vs the current refactored code https://github.com/PRBonn/kiss-icp/blob/v1.2.2/cpp/kiss_icp/core/Preprocessing.hpp

> Is it possible that the ouster mapping part introduces a different behavior?

I believe it is possible. I need to look deeper into it, but reverting the deskew point to 0.5 immediately fixed the issue on our side.

benemer, 2025-03-27:
Yes, please double-check. I now generated the maps using the ouster-sdk and can confirm what you observe. Since it does not happen using KISS alone, I suppose this comes from the ouster-sdk mapping, which requires further investigation.

I noticed that you do not integrate the de-skewed scan into the map but the raw one, is that correct?

Samahu, 2025-03-27:
> I noticed that you do not integrate the de-skewed scan into the map but the raw one, is that correct?

No, we don't generate a map, we write per column poses into the LidarScan which -currently- uses a simple interpolation between the updated pose & last pose.

I will look deeper into the issue and why this happen. If you guys don't think a parameter to control the deskew point is needed then feel free to close the ticket.

benemer, 2025-03-28:
> a simple interpolation between the updated pose & last pose.

This is most likely the issue.

When searching for this in the ouster-sdk, I cam across [this](https://github.com/ouster-lidar/ouster-sdk/blob/c8b9b4bd870a2c1a37e57ba5bd6df47c478f1141/python/src/ouster/sdk/mapping/util.py#L88) line in the `writeScanColPose` method:

```python
ts = normalized_ts_list[idx] - 0.5
```

Here you use a hardcoded timestamp of 0.5 as de-skewing reference. When changing that to 1.0, the final map looks a lot better.

In general, be aware that de-skewing after pose estimation using the estimated poses is not straightforward. The two poses you use for a second de-skewing represent the vehicle's pose at different points in time, depending on which timestamp you used in the first de-skewing as a reference.

For example, if you de-skew at normalized time 0.5, your poses represent the pose of the sensor in the middle betwen the beginning and end of the scan. If you now use your post-estimation de-skewing, you compute the motion between the middle of one scan duration and the middle of the next. Using this to interpolate the motion of a scan from the beginning to the end of the scan duration is not accurate, as discussed in the other threads here about de-skewing.

When changing the reference timestamp from 0.5 to 1.0, our estimated poses represent the pose of the sensor at the end of the scan. Now, you can correctly take a pose of a scan and its previous pose, compute the delta, and use it for de-skewing because this transformation exactly describes the motion from the beginning to the end of the scan.

Samahu, 2025-03-28:
This is one of the first thing that I tested actually, I did try to match the same deskew point with the interpolation point, the numbers but it doesn't get to the same level we had before:

Here is a summary of the result with each change:

| deskew point | interpolation point | pcd size (mb) | comment
|----------|----------|----------|----------|
| 1.0 (kiss-1.2+)  | 0.5 (unchanged) | 74M | worst outcome |
| 1.0 (kiss-1.2+) | 1.0 (your test) | 73M | better but not as good | 
| 0.5 (kiss-1.1) | 0.5 (unchanged) | 72M | baseline (ouster-sdk 0.14) |

This using the same example dataset from before `OS0-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap`, which has about 200 frames. You can reproduce the results by 1) running unmodified `ouster-cli(0.14)` and 2-3) later use a modified version and editing the interpolation point same as you did and run the following command for each time:

```bash
ouster-cli source OS0-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap slam save sample-${deskew-setting}.pcd
```

Note, a 1mb file size difference may not seem like a lot but the larger the file the more file differences are pronounced.

benemer, 2025-03-31:
Okay, that's interesting. Mainly because, as I mentioned above, the interpolation is not 100% correct when using a de-skewing point of 0.5.

You could reproduce my experiments above by accumulating the scans right after estimation to see whether the ouster-sdk mapping part is the issue.

Samahu, 2025-04-08:
My colleague has been looking into the issue on our end and will share his findings with you soon.

Samahu, 2025-04-13:
After some examination .. I think the issue is lies within the ouster-sdk iteslf and how we perform the interpolation and export the map. The changes made to the deskew point although affects our results, it isn't a KissICP issue. Closing the ticket.

closed 2025-04-13 by Samahu