PRBonn/kiss-icp #452: A problem met in wsl2, but sovled and as a reminder

runjtu, 2025-03-27:
Thx for your great work! I happened to meet an error named "PermissionError: [Errno 1] Operation not permitted: '/mnt/f/DataSets/kitti/bag/results/2025-03-27_22-23-41' ->
'/mnt/f/DataSets/kitti/bag/results/latest' " 
when I tried to run kiss icp in wsl2 ubuntu20.04

and the solution is to avoid generating results/* in the directory of /mnt. Just run commands in /home/usrname.

[image]

benemer, 2025-03-27:
You probably do not have read/write permissions to that folder. Can you post the output of `ls -lh /mnt/f/DataSets/kitti/bag' ?

runjtu, 2025-03-27:
Thanks and it is as follows
[image]

benemer, 2025-03-27:
Okay, that looks fine. Is the data on an external hard drive? Maybe the filesystem does not allow to create the symbolic link between the `latest` folder and the actual `results` folder.

You can still run it by cloning this repo and applying the following patch:

```diff
diff --git a/python/kiss_icp/pipeline.py b/python/kiss_icp/pipeline.py
index 987a87c..5446695 100644
--- a/python/kiss_icp/pipeline.py
+++ b/python/kiss_icp/pipeline.py
@@ -210,8 +210,6 @@ class OdometryPipeline:
         results_dir = os.path.join(os.path.realpath(out_dir), get_current_timestamp())
         latest_dir = os.path.join(os.path.realpath(out_dir), "latest")
         os.makedirs(results_dir, exist_ok=True)
-        os.unlink(latest_dir) if os.path.exists(latest_dir) or os.path.islink(latest_dir) else None
-        os.symlink(results_dir, latest_dir)
         return results_dir
 
     def _create_output_dir(self):
```

and then build KISS-ICP. This will remove the symlink step.

closed 2025-03-28 by runjtu