PRBonn/kiss-icp #453: How do I get the timestamp in kitti raw_data?

taoziyyy, 2025-03-31:
The kitti raw_data I downloaded has no timestamp information for each point cloud in each frame, how did the author handle this? Has anyone else encountered the same problem as me?

benemer, 2025-03-31:
To use KISS-ICP with KITTI raw and de-skewing, you need to specify the `kitti_raw` dataloder when running it.

This is how we do it for KITTI:

```python
    @staticmethod
    def get_timestamps(points):
        x = points[:, 0]
        y = points[:, 1]
        yaw = -np.arctan2(y, x)
        timestamps = 0.5 * (yaw / np.pi + 1.0)
        return timestamps
```

This is because KITTI uses a rotating LiDAR scanner, in which each column corresponds to the same timestamp.

taoziyyy, 2025-03-31:
How do I specify the kitti_raw dataloader? Like this: ros2 launch kiss_icp odometry.launch.py topic:=/kitti/velo/pointcloud dataloader:=kitti_raw

benemer, 2025-03-31:
Oh I see, you are using ROS. KITTI raw provides the data as binary files, so you can not use ROS here. You should use the Python API we provide.

closed 2025-04-14 by benemer