PRBonn/kiss-icp #454: pip install exits in CMake build stage due to missing directory in package

mendrzik, 2025-03-31:
When installing python package via pip (command: pip install kiss-icp --verbose), cmake complains that the subdirectory "kiss_icp/pybind" is missing:

[image]

When downloading the package which pip uses for installing the package I get the following files and directory as part of the tar.gz
[image]

Hence the directory is in fact missing.

- pip version: 25.0.1 
- python version: 3.12
- conda version: 25.1.1
- platform: arm64 (nvidia orin)
- OS: L4T 5.15.148-tegra

tizianoGuadagnino, 2025-04-03:
Hi @mendrzik, I don't understand are you trying to build the package from sources or just install it from pip? In the later case, what happen when you run `pip install kiss-icp`?

riadkudya-star, 2025-08-19:
###    ###__

LaVieEstDure, 2025-08-26:
Having the same issue on a Jetson Orin base
```
(slam) root@Pandora:/ros2_ws/src/slam# pip install kiss-icp
Collecting kiss-icp
  Using cached kiss_icp-1.2.3.tar.gz (35 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting natsort (from kiss-icp)
  Using cached natsort-8.4.0-py3-none-any.whl.metadata (21 kB)
Requirement already satisfied: numpy in ./lib/python3.12/site-packages (from kiss-icp) (2.3.2)
Collecting pydantic>=2 (from kiss-icp)
  Using cached pydantic-2.11.7-py3-none-any.whl.metadata (67 kB)
Collecting pydantic-settings (from kiss-icp)
  Using cached pydantic_settings-2.10.1-py3-none-any.whl.metadata (3.4 kB)
Collecting pyquaternion (from kiss-icp)
  Using cached pyquaternion-0.9.9-py3-none-any.whl.metadata (1.4 kB)
Collecting rich (from kiss-icp)
  Using cached rich-14.1.0-py3-none-any.whl.metadata (18 kB)
Collecting tqdm (from kiss-icp)
  Using cached tqdm-4.67.1-py3-none-any.whl.metadata (57 kB)
Collecting typer>=0.6.0 (from typer[all]>=0.6.0->kiss-icp)
  Using cached typer-0.16.1-py3-none-any.whl.metadata (15 kB)
Collecting annotated-types>=0.6.0 (from pydantic>=2->kiss-icp)
  Using cached annotated_types-0.7.0-py3-none-any.whl.metadata (15 kB)
Collecting pydantic-core==2.33.2 (from pydantic>=2->kiss-icp)
  Using cached pydantic_core-2.33.2-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (6.8 kB)
Requirement already satisfied: typing-extensions>=4.12.2 in ./lib/python3.12/site-packages (from pydantic>=2->kiss-icp) (4.15.0)
Collecting typing-inspection>=0.4.0 (from pydantic>=2->kiss-icp)
  Using cached typing_inspection-0.4.1-py3-none-any.whl.metadata (2.6 kB)
Requirement already satisfied: click>=8.0.0 in ./lib/python3.12/site-packages (from typer>=0.6.0->typer[all]>=0.6.0->kiss-icp) (8.2.1)
Collecting shellingham>=1.3.0 (from typer>=0.6.0->typer[all]>=0.6.0->kiss-icp)
  Using cached shellingham-1.5.4-py2.py3-none-any.whl.metadata (3.5 kB)
Collecting markdown-it-py>=2.2.0 (from rich->kiss-icp)
  Using cached markdown_it_py-4.0.0-py3-none-any.whl.metadata (7.3 kB)
Collecting pygments<3.0.0,>=2.13.0 (from rich->kiss-icp)
  Using cached pygments-2.19.2-py3-none-any.whl.metadata (2.5 kB)
WARNING: typer 0.16.1 does not provide the extra 'all'
Collecting python-dotenv>=0.21.0 (from pydantic-settings->kiss-icp)
  Using cached python_dotenv-1.1.1-py3-none-any.whl.metadata (24 kB)
Collecting mdurl~=0.1 (from markdown-it-py>=2.2.0->rich->kiss-icp)
  Using cached mdurl-0.1.2-py3-none-any.whl.metadata (1.6 kB)
Using cached pydantic-2.11.7-py3-none-any.whl (444 kB)
Using cached pydantic_core-2.33.2-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (1.9 MB)
Using cached typer-0.16.1-py3-none-any.whl (46 kB)
Using cached rich-14.1.0-py3-none-any.whl (243 kB)
Using cached natsort-8.4.0-py3-none-any.whl (38 kB)
Using cached pydantic_settings-2.10.1-py3-none-any.whl (45 kB)
Using cached pyquaternion-0.9.9-py3-none-any.whl (14 kB)
Using cached tqdm-4.67.1-py3-none-any.whl (78 kB)
Using cached annotated_types-0.7.0-py3-none-any.whl (13 kB)
Using cached markdown_it_py-4.0.0-py3-none-any.whl (87 kB)
Using cached pygments-2.19.2-py3-none-any.whl (1.2 MB)
Using cached python_dotenv-1.1.1-py3-none-any.whl (20 kB)
Using cached shellingham-1.5.4-py2.py3-none-any.whl (9.8 kB)
Using cached typing_inspection-0.4.1-py3-none-any.whl (14 kB)
Using cached mdurl-0.1.2-py3-none-any.whl (10.0 kB)
Building wheels for collected packages: kiss-icp
  Building wheel for kiss-icp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for kiss-icp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [61 lines of output]
      WARNING: Use cmake.version instead of cmake.minimum-version with scikit-build-core >= 0.8
      WARNING: Use build.verbose instead of cmake.verbose for scikit-build-core >= 0.10
      *** scikit-build-core 0.11.6 using CMake 3.28.3 (wheel)
      *** Configuring CMake...
      loading initial cache file build/cp312-cp312-linux_aarch64/CMakeInit.txt
[... 40 lines cut ...]
      CMake Deprecation Warning at build/cp312-cp312-linux_aarch64/_deps/tessil-src/CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 3.5 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      
      
      CMake Error at CMakeLists.txt:48 (add_subdirectory):
        add_subdirectory given source "kiss_icp/pybind" which is not an existing
        directory.
      
      
      -- Configuring incomplete, errors occurred!
      
      *** CMake configuration failed
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for kiss-icp
Failed to build kiss-icp
ERROR: Could not build wheels for kiss-icp, which is required to install pyproject.toml-based projects
```

@tizianoGuadagnino Is there any reason the jetson/ARM build process would not be supported / might be broken?