PRBonn/kiss-icp #458: Questions about the accuracy of deskew

hahayuxiang, 2025-04-11:
### Question about `relative_motion` angle behavior in normal driving scenarios

First of all, thank you for your excellent work!  
I'm currently trying to incorporate KISS-ICP into our engineering system design, and I’ve encountered a few issues that I would like to ask for your insights on.

According to the KISS-ICP paper:

> _"The constant velocity model assumes that a robot moves with the same translational and rotational velocity as in the previous time step."_

Based on this, I tried printing out the **rotation angle of `relative_motion`** using:

```cpp
double rotation_angle_rad = omega.head<3>().norm();
double rotation_angle_deg = rotation_angle_rad * (180.0 / M_PI);
...
std::ofstream out("/home/hahayuxiang/E/output/rotation_angle.txt", std::ios::app);
if (out.is_open()) {
    out << "Rotation angle (radians): " << rotation_angle_rad << "\n";
    out << "Rotation angle (degrees): " << rotation_angle_deg << "\n";
    out << "\n";
    out.close();
 } else {
      std::cerr << "Failed to open file for writing.\n";
 }
```

#### Sample output
````java
Rotation angle (radians): 1.05874
Rotation angle (degrees): 60.6614

Rotation angle (radians): 1.06265
Rotation angle (degrees): 60.8855

Rotation angle (radians): 1.05757
Rotation angle (degrees): 60.5944

Rotation angle (radians): 1.09086
Rotation angle (degrees): 62.5017

Rotation angle (radians): 1.0836
Rotation angle (degrees): 62.086

...

Rotation angle (radians): 0.0167852
Rotation angle (degrees): 0.96172

Rotation angle (radians): 0.0130171
Rotation angle (degrees): 0.745827

Rotation angle (radians): 0.00563802
Rotation angle (degrees): 0.323035

Rotation angle (radians): 0.016989
Rotation angle (degrees): 0.973396

Rotation angle (radians): 0.00696401
Rotation angle (degrees): 0.399008

...

Rotation angle (radians): 0.445457
Rotation angle (degrees): 25.5228

Rotation angle (radians): 0.441389
Rotation angle (degrees): 25.2897

Rotation angle (radians): 0.484867
Rotation angle (degrees): 27.7808

Rotation angle (radians): 0.490828
Rotation angle (degrees): 28.1224

Rotation angle (radians): 0.491076
Rotation angle (degrees): 28.1366

````
#### My observations

- During some intervals, the angle is close to 0°.

- However, in most frames, the angle stays within the range of 30° to 60° per frame.

- This behavior persists over long sequences (i.e., many consecutive frames), which seems unusual under typical vehicle motion.
#### My question

- In my understanding, relative_motion should represent the transformation from the previous frame to the current one. For a normally driving vehicle on the road, I would expect the inter-frame rotation angle to mostly stay under 20°. Large inter-frame rotations above 30° per 0.1s frame duration seem unlikely in such cases.
- Is this behavior expected when using last_delta_ for deskew?
- Could these large angles indicate a possible overestimation or instability in the motion model?
- Is there a recommended way to detect and handle such cases where relative_motion might not reflect realistic motion?
#### I am using the Mulran DCC2 dataset for testing.

### Thanks again for your amazing work!

benemer, 2025-04-14:
I think you are looking at the translational part of the correction. From the Sophus doc:

> The first three components of a represent the translational part upsilon in the tangent space of SE(3), while the last three components of a represents the rotation vector omega. To be more specific, this function computes expmat(hat(a)) with expmat(.) being the matrix exponential and hat(.) the hat-operator of SE(3), see below.

hahayuxiang, 2025-04-14:
Thank you for the clarification and for promptly pointing out my misunderstanding. 🙏  
Your explanation helped me see where I went wrong.

After making the necessary adjustments, I was able to get the expected results — everything works as intended now.  
Really appreciate your patience and the great support!

Best regards,  
Yuxiang

closed 2025-04-14 by hahayuxiang