PRBonn/kiss-icp #461: ROS2 Build ERROR

Dejavu-h, 2025-04-18:
```
lue, void>::type* <anonymous> > void rclcpp::AnySubscriptionCallback<MessageT, Alloc>::set(CallbackT) [with CallbackT = CallbackT; typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::shared_ptr<_Tp>)> >::value>::type* <anonymous> = <enumerator>; MessageT = const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&; Alloc = std::allocator<void>]’
   83 |   void set(CallbackT callback)
      |        ^~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:83:8: note:   template argument deduction/substitution failed:
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:81:17: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
   81 |     >::type * = nullptr
      |                 ^~~~~~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:97:8: note: candidate: ‘template<class CallbackT, typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::shared_ptr<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>, const rclcpp::MessageInfo&)> >::value, void>::type* <anonymous> > void rclcpp::AnySubscriptionCallback<MessageT, Alloc>::set(CallbackT) [with CallbackT = CallbackT; typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::shared_ptr<_Tp>, const rclcpp::MessageInfo&)> >::value>::type* <anonymous> = <enumerator>; MessageT = const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&; Alloc = std::allocator<void>]’
   97 |   void set(CallbackT callback)
      |        ^~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:97:8: note:   template argument deduction/substitution failed:
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:95:17: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
   95 |     >::type * = nullptr
      |                 ^~~~~~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:139:8: note: candidate: ‘template<class CallbackT, typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::unique_ptr<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&, std::default_delete<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&> >)> >::value, void>::type* <anonymous> > void rclcpp::AnySubscriptionCallback<MessageT, Alloc>::set(CallbackT) [with CallbackT = CallbackT; typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::unique_ptr<MessageT, typename std::conditional<std::is_same<typename std::allocator_traits<typename std::allocator_traits<_NodeAlloc>::rebind_traits<MessageT>::allocator_type>::rebind_alloc<MessageT>, typename std::allocator<void>::rebind<_Tp1>::other>::value, std::default_delete<_Tp>, rclcpp::allocator::AllocatorDeleter<typename std::allocator_traits<_NodeAlloc>::rebind_traits<MessageT>::allocator_type> >::type>)> >::value>::type* <anonymous> = <enumerator>; MessageT = const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&; Alloc = std::allocator<void>]’
  139 |   void set(CallbackT callback)
      |        ^~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:139:8: note:   template argument deduction/substitution failed:
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:137:17: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
  137 |     >::type * = nullptr
      |                 ^~~~~~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:153:8: note: candidate: ‘template<class CallbackT, typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::unique_ptr<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&, std::default_delete<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&> >, const rclcpp::MessageInfo&)> >::value, void>::type* <anonymous> > void rclcpp::AnySubscriptionCallback<MessageT, Alloc>::set(CallbackT) [with CallbackT = CallbackT; typename std::enable_if<rclcpp::function_traits::same_arguments<CallbackT, std::function<void(std::unique_ptr<MessageT, typename std::conditional<std::is_same<typename std::allocator_traits<typename std::allocator_traits<_NodeAlloc>::rebind_traits<MessageT>::allocator_type>::rebind_alloc<MessageT>, typename std::allocator<void>::rebind<_Tp1>::other>::value, std::default_delete<_Tp>, rclcpp::allocator::AllocatorDeleter<typename std::allocator_traits<_NodeAlloc>::rebind_traits<MessageT>::allocator_type> >::type>, const rclcpp::MessageInfo&)> >::value>::type* <anonymous> = <enumerator>; MessageT = const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&; Alloc = std::allocator<void>]’
  153 |   void set(CallbackT callback)
      |        ^~~
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:153:8: note:   template argument deduction/substitution failed:
/opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:151:17: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
  151 |     >::type * = nullptr
      |                 ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /usr/include/eigen3/Eigen/Core:96,
                 from /debug/demo/kiss-icp/ros/src/OdometryServer.cpp:23:
/usr/include/c++/9/ext/new_allocator.h: In instantiation of ‘class __gnu_cxx::new_allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>’:
/usr/include/c++/9/bits/allocator.h:112:11:   required from ‘class std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>’
/usr/include/c++/9/bits/shared_ptr_base.h:1365:63:   required by substitution of ‘template<class _Yp> struct std::__shared_ptr<std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>, __gnu_cxx::_S_atomic>::__has_esft_base<_Yp, std::__void_t<decltype (__enable_shared_from_this_base(declval<const std::__shared_count<__gnu_cxx::_S_atomic>&>(), declval<_Yp*>()))> > [with _Yp = std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>]’
/usr/include/c++/9/bits/shared_ptr_base.h:1387:21:   required by substitution of ‘template<class _Yp, class _Yp2> typename std::enable_if<(! std::__shared_ptr<std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>, __gnu_cxx::_S_atomic>::__has_esft_base<_Yp2, void>::value), void>::type std::__shared_ptr<std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>, __gnu_cxx::_S_atomic>::_M_enable_shared_from_this_with<_Yp, _Yp2>(_Yp*) [with _Yp = std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>; _Yp2 = std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>]’
/usr/include/c++/9/bits/shared_ptr_base.h:1345:4:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&> >; _Args = {std::allocator<void>&}; _Tp = std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr.h:359:59:   [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/c++/9/bits/shared_ptr.h:717:39:   required from ‘std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = std::allocator<const std::shared_ptr<const sensor_msgs::msg::PointCloud2_<std::allocator<void> > >&>; _Args = {std::allocator<void>&}]’
[... cut ...]
  667 |       function<_Res(_ArgTypes...)>::
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/odometry_component.dir/build.make:76: CMakeFiles/odometry_component.dir/src/OdometryServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:226: CMakeFiles/odometry_component.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< kiss_icp [4.03s, exited with code 2]

Summary: 0 packages finished [4.15s]
  1 package failed: kiss_icp
  1 package had stderr output: kiss_icp

```

Dejavu-h, 2025-04-18:
Hello, I'm using ROS2 Foxy, Ubuntu 20.04, and I am getting error when building package.
and I use this method

Dejavu-h, 2025-04-18:
Hello, I'm using ROS2 Foxy, Ubuntu 20.04, and I am getting error when building package.
and I use this method #171

benemer, 2025-04-23:
What exactly do you mean with 

> I use this method #171 

?

Dejavu-h, 2025-04-23:
Hello, I have solved the problem, I modified the callback function in the OdometryServer.cpp file.

benemer, 2025-04-23:
Can you please post the fix here for future reference? Even though Foxy has reached its end-of-life, others may also struggle with this. Thanks!

smallwat3r, 2025-10-02:
I also came across this issue when using Foxy. 

Here is what I changed in `OdometryServer.cpp`:

```
void OdometryServer::RegisterFrame(sensor_msgs::msg::PointCloud2::ConstSharedPtr msg_ptr) {
    const auto& cloud = *msg_ptr;  

    const auto cloud_frame_id = cloud.header.frame_id;
    const auto points         = PointCloud2ToEigen(msg_ptr);
    const auto timestamps     = GetTimestamps(msg_ptr);

    const auto& [frame, keypoints] = kiss_icp_->RegisterFrame(points, timestamps);
    const Sophus::SE3d kiss_pose   = kiss_icp_->pose();

    PublishOdometry(kiss_pose, cloud.header);
    if (publish_debug_clouds_) {
        PublishClouds(frame, keypoints, cloud.header);
    }
}
```

Also, don't forget to update `OdometryServer.hpp`:

```
    void RegisterFrame(const sensor_msgs::msg::PointCloud2::ConstSharedPtr msg_ptr);
```